prusa3d/PrusaSlicer #13902: Sequential Printing: Brm

Elarion245, 2025-01-04:
### Description of the bug

I usually use sequential printing for multiple objects. To combat warping I have added a brim in Prusa 2.9, which works as expected.
I have noticed though that in the first layer of my print, which contains two objects, both brims are printed first, followed by the first object and then the second.

My expected behavior would be that the brim corresponding to the second object is printed once the first object is finished.

### Project file & How to reproduce

Arbitrary Project, enable brim and sequential printing
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Win10

### Printer model

Lerdge iX

dhlavaty, 2025-03-22:
I can confirm, this is a very annoying bug. But it doesn't seem as a v2.9 bug. It is present in v2.7.4 as well. I didn't go further into the history.

neophyl, 2025-09-02:
Also if your first layer bed temperature is different it stops any requirement to let the bed temp reach target in between objects.   This one has been around forever, you can pretty much to back to https://github.com/prusa3d/PrusaSlicer/issues/292  (May 2017)

I also remember that there have been requests for the entire first layer for all objects to be printed first and only after the first layer has been done then switch to sequential printing.  So you know you have a good first layer for all objects.

Stoutlimb, 2026-01-04:
Tomorrow it's been 1 year without anything being assigned.  This bug DESTROYS my work flow.

For my 3d print to work correctly, it was necessary to turn off retraction.  (The retraction between brim and start of model killed my adhesion.)  Works GREAT for one model at a time.  With batch printing, it's an utter disaster unless the full model follows immediately after the brim.  Can you imagine a 0.8 nozzle slobbering all over the build plate as it moves to the next fully made brim, then failing to adhere as the flow isn't there?

Please at least assign someone to this, I have to switch to a much less liked slicer because of this failure.

Elarion245, 2026-09-18:
It would be nice for this to be solved in 3.X