prusa3d/PrusaSlicer #13904: single layer part is ignored if its height is above 37.5mm

RPospisil, 2025-01-04:
### Description of the bug

I tested support so I created object with 2 parts (each of one layer of height 0.2mm) - one is on the bed, the second is above. 

I sliced with 0.20mm SPEED/STRUCTURAL and the 2nd part is in the height 37.6mm and higher then the part is ignored (not sliced), if it is below 37.6, then it is sliced. 

### Project file & How to reproduce

If I slice it with 0.15mm settings, then both parts are sliced.

What is more stranger, then if you open the provided 3mf file, then the 'above bed' part is never sliced regardless its height.

To reproduce it you have to create the project manually (you can be inspired by the provided 3mf)
1) select MK4S  0.4 nozzle printer, settings is 0.20mm SPEED and enable supports everywhere (or on build plate)
2) add box, uncheck the lock icon and set its height to 0.2mm 
3) duplicate the box and move it so the boxes are not overlapping
4) merge both boxes to one object

5) set height of one of the box part to 37.5mm
6) slice it - both boxes are sliced

7) set height of one of the box part to 37.6mm
8) slice it - box on the bed is sliced, but the second is missing

Screenshots

[image]

[image]

Project file
[single layer part is ignored issue.zip](https://github.com/user-attachments/files/18307648/single.layer.part.is.ignored.issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11 Pro

### Printer model

MK4S, XL

u89djt, 2025-01-05:
(Fellow user with a potential explanation to verify) Within each layer, extrusion is allocated if greater than 50% of the height of the extrusion is occupied by mass in the model. You've got a layer height of 0.2mm and half the height of a 0.2mm (actually 0.19999 but I think that doesn't change anything) object split into two of 0.1mm either side of a layer boundary. Since there's half or less of the volume of any extrusion occupied, no extrusions are generated.

RPospisil, 2025-01-05:
@u89djt I agree that in case of positioning the 0.2 height box on odd height, then it can 'disappear'. I've tested it more and there is approximation error (you mentioned) which causes it - in my case the 'above box' is sliced for heights 37.1, 37.2, 37.3, 37.4, 37.5. Above 37.6 it is sliced for odd-ended heights only  - i.e., 37.7, 37.9, 38.1 .... but not for 37.6, 37.8, 38.0 .

In fact it is really confusing, that if I put 0.2mm-height-object in 37mm height then it is sliced, but it is not sliced at 38mm.
There has to be warning that the object cannot be (or won't be) sliced due to approximation error.... 

Can you elaborate more on your explanation? Is it some math approximation error (as you mentioned, that 0.2mm is in reality 0.19999mm)?  Is it due to number representation format?

u89djt, 2025-01-05:
When you say "put it in/at a height" are you referring to the bottom of the object, the top of the object, or the middle of the object? When you type a number in to the Z location box, I think it will set the location of the centre of the object. When you type 0.1 or 0.2 into the Z offset box for a move command, it will move the whole object by that quantity. It could be easy to confuse where the object actually is.
I speculate at this point that it probably doesn't matter that it's 0.1999 rather than 0.2.

RPospisil, 2025-01-05:
By "part height" I'm referring to the "Position Z" value of the part in the object - so it is the z position of part's middle. I do not mean the height of the whole object (see the project 3mf file and "How to reproduce").

Anyway It is strange that if I load the 3mf file, then prusa slicer does not slice the 'part above' even if its height its below 37.6 ( see above).

u89djt, 2025-01-05:
It's behaving the way I expect it to. I'm not seeing anything confusing, so I'd need to dig around a bit more to appreciate what you're seeing. I might get a chance to do that, but I invite other folk looking in to take over if they have time.
While you wait, have a look at this file:
[hit and miss.zip](https://github.com/user-attachments/files/18311198/hit.and.miss.zip)
[image]

RPospisil, 2025-01-05:
I opened your 3mf project and it looks different from your screenshot - the box with part above at height 37.4 is sliced, while in your case it is not.
For you project I'm getting errors - both caused by not enabled supports.
- empty layer between 0.2 and 37.4 - refers to 37.4 object which shall be printable but with supports. The other 37.6 object is ok as it ignores the box part at height 37.6.
- detected printed stability issues - it is expected as supports are not enabled

Further it seems that there is some issue in save and open 3mf project file as it does not behave consistently....  

[image]

u89djt, 2025-01-05:
OK, there's something here: when I open your original file and slice it, I get this:
[image]
I then close the slicer and double click on your original file again, then without slicing (to avoid generating any information somehow),
"save as" immediate.3mf, close the slicer, double click on immediate.3mf and slice and get this:
[immediate.zip](https://github.com/user-attachments/files/18311614/immediate.zip)
[image]

closed 2026-09-29 by prusaslicer-bot[bot]