prusa3d/PrusaSlicer #13906: Fresh install of Prusa Slicer 2.9.0 crashing when opening the config wizard or clicking on the printables tab

Herrcookiekiller, 2025-01-04:
### Description of the bug

Prusa Slicer 2.9.0 crashes without any error message when either
_•clicking on the Printables tab_
or
_•opening the configuration wizard_

The only thing i could find was this error report filed by windows, as well as an Event viewer Error log
Both attached below.

```
Version=1
EventType=APPCRASH
EventTime=133805053585510729
ReportType=2
Consent=1
UploadTime=133805053590915606
ReportStatus=268435456
ReportIdentifier=06e8c49b-ea40-4875-8e2b-cc7a5cacbbc1
IntegratorReportIdentifier=08409678-1dfd-47a8-b1cf-7cca3041a0aa
Wow64Host=34404
NsAppName=prusa-slicer.exe
OriginalFilename=prusa-slicer.exe
AppSessionGuid=00001d5c-0007-0046-7551-1ab5fc5edb01
TargetAppId=W:0006a3f8810e7f9bf075d201adb035b0906c00000904!0000b986a8adfdd0beb79fffdc347d236658f9403447!prusa-slicer.exe
TargetAppVer=2024//12//20:13:04:17!37bb3!prusa-slicer.exe
BootId=4294967295
TargetAsId=5600
UserImpactVector=808452912
IsFatal=1
EtwNonCollectReason=1
Response.BucketId=b0d2ad73946ded264baa5e2cd3f81de1
Response.BucketTable=4
Response.LegacyBucketId=1993509331692690913
Response.type=4
Sig[0].Name=Anwendungsname
Sig[0].Value=prusa-slicer.exe
Sig[1].Name=Anwendungsversion
Sig[1].Value=2.9.0.0
Sig[2].Name=Anwendungszeitstempel
Sig[2].Value=67656b51
Sig[3].Name=Fehlermodulname
Sig[3].Value=EmbeddedBrowserWebView.dll
Sig[4].Name=Fehlermodulversion
Sig[4].Value=121.0.2277.83
Sig[5].Name=Fehlermodulzeitstempel
Sig[5].Value=65b1bfd9
Sig[6].Name=Ausnahmecode
Sig[6].Value=c000041d
Sig[7].Name=Ausnahmeoffset
Sig[7].Value=00000000003450d0
DynamicSig[1].Name=Betriebsystemversion
DynamicSig[1].Value=10.0.19045.2.0.0.256.48
DynamicSig[2].Name=Gebietsschema-ID
DynamicSig[2].Value=1031
DynamicSig[22].Name=Zusatzinformation 1
DynamicSig[22].Value=5383
DynamicSig[23].Name=Zusatzinformation 2
DynamicSig[23].Value=538350a08bff3e191f4238b9fba4ded7
DynamicSig[24].Name=Zusatzinformation 3
DynamicSig[24].Value=7bdb
DynamicSig[25].Name=Zusatzinformation 4
DynamicSig[25].Value=7bdb65c1bf1df689bdd79a253bc483cc
UI[2]=C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
LoadedModule[0]=C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
LoadedModule[1]=C:\WINDOWS\SYSTEM32\ntdll.dll
LoadedModule[2]=C:\WINDOWS\System32\KERNEL32.DLL
LoadedModule[3]=C:\WINDOWS\System32\KERNELBASE.dll
LoadedModule[4]=C:\WINDOWS\System32\USER32.dll
LoadedModule[5]=C:\WINDOWS\System32\win32u.dll
[... 231 lines cut ...]
```

### Project file & How to reproduce

1. Install Prusa Slicer with the .exe from https://www.prusa3d.com/page/prusaslicer_424/
2. Run the installed exe
3. Click on add printers or profiles
4. Watch Prusa Slicer crash

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 10 Pro 22H2

### Printer model

MK4S

wobblebobble, 2025-01-08:
Hi, 
(almost) Same here.
I´m new to the Prusa Slicer and installed 2.9.0 on my Windows 11 Machine, as I have received my new PrusaXL
The program crashes after startup directly after bulding up the GUI, but no wizard is displayed before that.

Herrcookiekiller, 2025-01-08:
> Hi, (almost) Same here. I´m new to the Prusa Slicer and installed 2.9.0 on my Windows 11 Machine, as I have received my new PrusaXL The program crashes after startup directly after bulding up the GUI, but no wizard is displayed before that.

If you dont see the configuration wizard trying to load in a seperate window just before Prusa Slicer crashes, i dont think this is related.

kocikdav, 2025-01-09:
Hello @Herrcookiekiller thank you for the report. Would you please help us understand your problem by pasting us the console log?
How to: Open directory with PS installation in command line. Run `./prusa-slicer-console.exe --loglevel=5`
There should be some information in the log, that hopefully help us to reproduce this issue and fix it.

Herrcookiekiller, 2025-01-09:
> Hello @Herrcookiekiller thank you for the report. Would you please help us understand your problem by pasting us the console log? How to: Open directory with PS installation in command line. Run [...]

Hey, thanks for responding!
Yesterday i looked at the logs i already posted _again_ and noticed:

`
LoadedModule[92]=C:\Program Files\Prusa3D\PrusaSlicer\WebView2Loader.dll
LoadedModule[93]=C:\Program Files (x86)\Microsoft\EdgeWebView\Application\121.0.2277.83\EBWebView\x64\EmbeddedBrowserWebView.dll
`
in the Error Report

as well as

`EmbeddedBrowserWebView.dll `
in the Event Viewer Log.

Specifically the part  \EdgeWebView\ was quite suspicious.
This made me come to the conclusion that there must be something wrong with a certain _Webbrowser_ i may or may not have installed. I did not have Edge installed.

Now i assume that you guys are using some of Microsoft Edges DLL's or parts of the browser to render the Printables Tab and the Configuration Wizard.

I installed Edge, activated all Microsoft updates in Winaero Tweaker and launched a few previous versions (mainly 2.7.1 and 2.8.0) from the unpacked zip folder of each Prusa Slicer version you can download here in the releases section to generate some user-profile files that get stored under /AppData/Local as far as i know.

Today i launched the 2.9.0 version from within the zip folder and it came up with the configuration wizard, the Printables tab worked as well.

After a full install with the prusa3d_installer.exe from your [PrusaSlicer Download Page](https://www.prusa3d.com/page/prusaslicer_424/), everything now works as expected.

I ran the now working version with your suggested loglevel command anyways. [Heres the pastebin](https://pastebin.com/raw/teXC33us).

**For anyone interested, when installing Edge select "install for local user only" this worked for me.**

lonnon, 2025-01-26:
I get exactly the same crash behavior in 2.9.0 on Fedora Linux 40, running from the official flatpak installation. Clicking the Printables tab or running the Configuration Wizard both result in a crash to desktop.

I don't know how Prusa Slicer runs behind the scenes on Linux (WINE or another Windows emulation layer?), but installing Edge as a workaround is obviously not going to work on my system. I'm happy to provide logging information if you can tell me where to look for it on linux.

hiinaspace, 2025-04-13:
I get the same crash, on printables browser and prusalink login, using 2.9.2 from flatpak on ubuntu 24.04. The last message in the logs at debuglevel=5 is `open login browser: https://account.prusa3d.com/o/authorize/?embed=1&client_id=oamhmhZez7opFosnwzElIgE2oGgI2iJORSkw587O&response_type=code&code_challenge=TL2qYnq6NrxOydTAAHZ5QTRZP2hTQCgSIdAfNcRC3Iw&code_challenge_method=S256&scope=basic_info&redirect_uri=prusaslicer://login&language` . no indication as to what's wrong with the browser however.

SachCZ, 2025-04-14:
Hello @hiinaspace, can you please share with us what graphics card are you using?

hiinaspace, 2025-04-14:
I have `AMD Ryzen 7 7840U w/ Radeon  780M Graphics` on a laptop. I'm also running wayland with the sway compositor, which might be somewhat unusual. FWIW, the 2.8.0 appimage still works on my machine, including the prusalink login.

SachCZ, 2025-04-14:
Thanks @hiinaspace, we are trying to collect us much feedback as possible, because we cannot really reproduce the issue.

Yigte, 2025-05-21:
If you’ve used tools like WinUtil, O&O ShutUp10++, or made changes that affect Microsoft Edge or Edge WebView2, you may encounter errors like "failed to load web browser" in some applications too.

This issue is often caused by the missing WebView2 Runtime, which many apps rely on to display embedded web content.

Fix 1 - Install WebView2 Runtime:
If WebView2 is missing, you can download and install it from the following link:
[Microsoft Edge WebView2 Runtime (Evergreen Standalone Installer)](https://developer.microsoft.com/en-us/microsoft-edge/webview2/)

Fix 2 – Repair system files:
System integrity might have been affected. Here's how to repair it:

Open the Start Menu, type cmd.
Right-click on Command Prompt and select Run as administrator.
Enter the following commands one by one, pressing Enter after each:
```
DISM /Online /Cleanup-Image /RestoreHealth
sfc /scannow
```

These commands check and repair missing or corrupted system components.
After the process finishes, restart your computer and try launching the application again.

This is how I solved the issue on my system.

closed 2026-09-29 by prusaslicer-bot[bot]