prusa3d/PrusaSlicer #13907: Export name sometimes contain variable names

Tony763, 2025-01-05:
### Description of the bug

When exporting sliced bgcode name of file sometimes contain variable names instead of actual materials/time: "Bridge v9_0.4n_0.2mm_{printing_filament_types}_MK4SMMU3_{print_time}.bgcode"

### Project file & How to reproduce

Slice models and hit an export button:

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0 Flatpack

### Operating system

Ubuntu 24.10

### Printer model

MK4 MMU3

Jan-Soustruznik, 2025-01-09:
Hello, @Tony763 , could you share some detailed info or video of the issue, please? I cannot reporoduce. Thank you

godlike64, 2025-03-11:
This issue is still present on 2.9.1 albeit it's tricky to reproduce. Just downloaded the flatpak. Steps to reproduce on PS configured for a 5T XL:

1. Download https://www.printables.com/model/367033-never-lost-ikea-samla-clip-remix-11l-22l-45l-65l and add `45-65l clip v1.stl` to the plate.
2. Cut the model in half horizontally on the default position.
3. Copy the two resulting models once.
4. Copy one of the 4 models once for a total of 5 pieces.
5. Set "Complete individual objects" in Print Settings.
6. Hit 'A' for automatic rearrangement.
7. Moved them to lower right cuadrant (they occupy 4 bed tiles).
8. Hit Slice.
9. Hit Export G-Code

At this point, the resulting filename had "{printing_filament_types}" in the name in place of "PLA". The strange thing was that I cancelled the export, then hit the button again and it was generated with the correct filename. Maybe there's a race condition somewhere?. I tried to re-slice and re-export but the issue did not reoccur.

lukasmatena, 2025-03-11:
This is actually not a bug. PrusaSlicer allows to "schedule" saving of the file before the export has finished. In that case, the filename in the dialog contains unexpanded placeholders, because the values are not known yet. If you confirm the dialog, the file on the disk will have the placeholders expanded, because the actual file is always exported after all operations are finished. All that is needed to "reproduce" it is to click Export button while slicing is still in progress.

Unless it is something else, I suggest to close this issue.

Tony763, 2025-03-11:
If it is "scheduled", shouldn't it wait for it's completion and properly fill values upon saving when there are placeholders in name? 🤔

lukasmatena, 2025-03-12:
It does wait before it exports the file, and fills the values correctly when saving it. But the user has to enter the filename when "scheduling", otherwise there is no point in scheduling the export.
I am not saying that this "scheduling" feature is a must have, and you are not the first person who got confused. I am just describing current behavior.

closed 2026-09-29 by prusaslicer-bot[bot]