prusa3d/PrusaSlicer #13915: Print_time and used_filament placeholder usage in custom G-code

Cheesemuffinz, 2025-01-06:
Creality printers display the estimated print time, filament used and layer height when selecting a print. This functionality requires those values to be present in the G-code file in such format:

Example (from Cura)
```
;TIME:2582
;Filament used: 1.57225m
;Layer height: 0.2
```

Prusa slicer does not emit these values into the G-code and custom G-code only allows the use of "layer_height" and does not allow the **existing** placeholders "print_time" and "used_filament" to be used in custom G-code. Please allow the usage of these placeholders in custom G-code.

Example
```
;TIME:{print_time}
;Filament used: {used_filament}
;Layer height: {layer_height}
```

Possibilities that come along with allowing the use of these placeholders are not limited to the one presented here.

neophyl, 2025-01-06:
They might not be at the **start** or match Cura (why should it be identical to some other software anyway ? ) but at the bottom of a sliced file is information.  For example from a random gcode I had sliced -
; objects_info = {"objects":[]}
; filament used [mm] = 2243.91
; filament used [cm3] = 5.21
; filament used [g] = 6.47
; filament cost = 0.19
; total filament used [g] = 6.47
; total filament cost = 0.19
; total filament used for wipe tower [g] = 0.00
; estimated printing time (normal mode) = 47m 27s
; estimated printing time (silent mode) = 47m 31s
; estimated first layer printing time (normal mode) = 1m 53s
; estimated first layer printing time (silent mode) = 1m 54s

So that has filament used and print time.  
Also every sliced file has ALL the settings used at the bottom too so looking through those  we have amongst them -
; first_layer_height = 0.2
; layer_height = 0.15

So more information than what Cura provides.  It's all in there if someone wants to parse the contents of the file.  Its just as easy for software to parse stuff from the end of a file as the start, makes no difference to code.

Cheesemuffinz, 2025-01-06:
True, it is possible with post processing scripts to move the needed parts to the top. 
However, I think it would be quite a crucial functionality in a slicer to output the code in the proper format in the first place, which is specified per printer, in their profiles.
Then again 3D printing is kind of a DIY hobby so writing a post-prosessing script is part of the fun. We need to also keep in mind the people who cannot code.

neophyl, 2025-01-06:
My point is that the info is already part of the gcode file, so if creality wanted to display it they could. 
You could ask creality to support Prusa slicer, orca slicer, slic3r as they all have the same. But that’s not really in their interests is it as they want people to use their slicer Cura. 
Prusa already support different firmware flavours, marlin etc which creality use on their printers, it’s a reality’s choice what parts of marlin or klipper etc they implement.

Cheesemuffinz, 2025-01-08:
Cura is Ultimaker's slicer not Creality's, and it still works out of the box with Creality printers when right profile is selected. I think that it is the Slicers job to make the G-code into the correct format. Prusa Slicer even has profiles that you can configure for Creality printers but they do not display the said information, meaning that is leaves the job halfway done.

neophyl, 2025-01-08:
You are still missing the point.  Cura is Creality's recommended slicer for their printers, they have made **their** implementation of the open source firmware they use on their printers compatible with a Cura feature.  If they wanted **their** firmware to show that same information from a gcode file sliced with Slic3r/PS/Super Slicer/Orca then they could easily do so.

Meanwhile due to the already existing post processing function that is kindly built into those slicers it is possible to do what you want as **the information is already there in the file to be used**.  

This extra info that you want is NOT part of any gcode standard.  So much of the stuff included in m odern files is not part of any standard.  Is there a good case for a new standard, yeah.  There always is.  Good luck with that though.  Also the profiles supplied for non prusa printers are generally supplied by the user base and not prusa employees.  

This should be a feature request for Creality firmware devs.  

You could of course modify Prusa Slicers code to make it do what you want yourself and submit a pull request.  Put your own time and effort in to help the Creality user base out.   Personally I think using as Post Processing script will be easier.

I'd rather Prusa Dev's spend their limited time on one of the MANY other open issues that actually effect quality of prints rather than on supporting printer non essential stuff (that can already be done by the user), from a large company that should be fixing their own codebase to add support for a really popular family of slicers.

Personal opinion of course.  Hey your feature request might already be on their list, who knows.

mstetson, 2025-01-22:
I was just looking for this feature myself and decided to dig into the code a bit. It would take quite a bit of work for PrusaSlicer to support `print_time` and `used_filament` variables in the Start G-Code because those values are calculated during the export and so aren't known until after the Start G-Code has been written. You can see it in [src/libslic3r/GCode.cpp](https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/GCode.cpp) by looking at the usage of `update_print_estimated_stats` and `placeholder_parser_process`. I'm going to use a post-processor.

closed 2026-09-29 by prusaslicer-bot[bot]