prusa3d/PrusaSlicer #13923: Missing external perimeters after slicing

MickThePyro, 2025-01-07:
### Description of the bug

Sliced a part, and proceeded to print it. Did not notice this glitch when looking at the sliced object; only after the part was printed (so a waste of 5 hours time and a bunch of filament). There are some external perimeters that were clearly messed up. Examination of the sliced object showed the external perimeters were missing entirely, even though the finished print seems to show extra material beyond the intended surface. This appears on both the outside and inside of the part (one contiguous external perimeter).

I turned off Scarf Seams, and this issue went away. Clearly Scarf Seams are not ready for prime time yet. Good concept; just needs more work.

### Project file & How to reproduce

Slice file with included settings. Scarf Seams "everywhere" appears to be the cause of this problem.

[GCB Renishaw cover PRINT (glitch).zip](https://github.com/user-attachments/files/18326310/GCB.Renishaw.cover.PRINT.glitch.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11 64bit

### Printer model

Prusa MK4S

TerPuc, 2025-01-07:
Dear @MickThePyro, thank you for your post. We will have a look at this. Internal issue created: SPE 2639.
I would just like to ask out of curiosity - why do you use XY size compensation -0.003 mm? It seems like a rather negligible compensation. 
Thank you and have a nice day!

micklayden, 2025-01-07:
@TerPuc, that's an interesting observation. I did not intentionally set the XY compensation to -0.003mm. Is there some configuration where this is a default placeholder value?

TerPuc, 2025-01-08:
Hello @micklayden @MickThePyro, I am not aware about such configuration. However if you change it back to default value you should be able to print the cover without the defect you provided in the picture. In the meantime we will take a look at why this specific combination in settings caused this issue and we will work on the fix 👍

Jan-Soustruznik, 2025-02-18:
Closed as solved by the new release of [2.9.1-Alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1)

closed 2025-02-18 by Jan-Soustruznik