prusa3d/PrusaSlicer #13926: Calculation of print timer and filament needed

KimA998, 2025-01-07:
### Description of the bug

You might consider to have a look on your time and filament calculations in PrusaSlicer 2.9.0 with MK3.9.
I have just made a scheduled 82 hours and 42 minutes print in 98 hours and 0 minutes - it´s 18,49% more time to complete.
Wipe tower calculation was 525 grams to 451,1 grams in real life.
The print had 5 colors (6th was added while printing) and 2792 filament changes - NO user interventions at all (and that was nice going PRUSA!!!!)
I have standard length on the PTFE tube from MMU to extruder. I have made manual purge volumes in the slicer.
[image]
[image]
[image]

### Project file & How to reproduce

Just hit Slice button!

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

WIndows 11

### Printer model

MK3.9 with MMU3

Jan-Soustruznik, 2025-01-09:
Hello, @KimA998 , could you share your project file(3MF), please? Thank you.

KimA998, 2025-01-09:
Hi Jan

Here it is!

Kim

Jan-Soustruznik, 2025-01-10:
@KimA998 , could you try it again. I cannot find any attachement. Maybe is neccessaty to add file thru github page.

neophyl, 2025-01-10:
Don't forget it needs zipping up.

KimA998, 2025-01-10:
The file is now on github!

Kim

Jan-Soustruznik, 2025-01-13:
Hello @KimA998, I still don't see the file. Could you please upload it to Dropbox or Google Drive and share the download link? It’s possible that the file is too large to be attached to the GitHub ticket. Thank you, have a nice day

KimA998, 2025-01-15:
Hi Jan

I have now again tried to put it on GitHub and SAVED my comment as you can see at the screendump.

Kim

[cid:0c0d3143-4a99-45f7-b5f8-2ca90d85892b]

Jan-Soustruznik, 2025-01-20:
Hello, @KimA998 , I noticed that you’re responding via email. Please be aware that GitHub doesn’t attach email attachments to tickets. Could you kindly send me the attachment directly to my email at jan.soustruznik@prusa3d.cz, or upload it through the web interface? Thank you, have a nice day

Jan-Soustruznik, 2025-02-20:
Hello @KimA998,

Sorry for the late reply—I was waiting for a free slot in our test print farm to reprint your project. Every completed print has the same wipe tower height as provided by the Slicer.

There should be a simple solution in your filament settings. You have 3DE filament profiles, but I didn’t find them in our system profiles, so there may have an incorrect material density setting. This could affect the weight calculation of the wipe tower.

Another issue we noticed is variations in print time, but they were within 2–5%, which I believe is acceptable. Do you know if your print was interrupted (e.g., due to a collision, auto-homing, etc.) during the process? That could explain these differences.

closed 2026-09-29 by prusaslicer-bot[bot]