prusa3d/PrusaSlicer #13927: Relocate "Printables" button and/or change its style

Tupson444, 2025-01-07:
Currently (2.9.0), `Printables` button is the same style and right next to `Plater`, `Print Settings`, `Filaments` and `Printers`, while not being related to the other tabs. It is too easy to accidentally click on the `Printables` instead of `Printers` (especially because the words are similar).

Please can it simply be adequately distinguished from the other tabs? Either relocate it or change its style (or both).

This issue is related to #13635, which asked to add a toggle for `Printables` tab. That issue is closed because "not everything should be customizable", so here I propose an alternative.

m00dawg, 2025-03-13:
I ended up forking PS and removing it myself. I too am very disappointed in Prusa over this, especially since it's pretty easy to hide. Took me maybe an hour to figure it out and I don't really do any UI programming. Seems to me having a config option would take someone who knew the code better very little time to implement. At any rate, hopefully this helps folks:

https://github.com/prusa3d/PrusaSlicer/commit/c8d2a4fcdecfac421b1128b35dbd76ed7232b462

lukasmatena, 2026-09-01:
Redesigned in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena