prusa3d/PrusaSlicer #13928: Plater displays an empty screen.

fjpanag, 2025-01-08:
### Description of the bug

My previously functional PS v2.9.0 installation is no longer working. The plater tab displays an empty screen.

This may (or may not) be related to a system update. I did updated my OS, but I cannot surely correlate this to this issue.

### Project file & How to reproduce

Project File: [Untitled.zip](https://github.com/user-attachments/files/18340043/Untitled.zip)

Screenshot of the issue:

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0 (Flatpak)

### Operating system

Fedora 41

### Printer model

Voron V2.4r2 - 350mm

Jan-Soustruznik, 2025-01-09:
Hello, @fjpanag , The issue you reported is very difficult to diagnose with the provided information. However, I wonder if the problem might be related to the graphics driver. Could you try installing the graphics driver via Flatpak?

A similar issue was addressed here https://github.com/prusa3d/PrusaSlicer/issues/13653#issuecomment-2567132082, which led me to the idea that this could be an error related to the graphics driver.

I hope that will help you. Let me know if you will have some news.

fjpanag, 2025-01-11:
@Jan-Soustruznik I updated again all flatpaks and now it is working... Maybe something was broken, or any dependency was outdated?

I am closing this now, and if appears again I will try to gather more information on this.

closed 2025-01-11 by fjpanag