prusa3d/PrusaSlicer #13929: There is an object with no extrusions in the first layer

tcurdt, 2025-01-08:
### Description of the bug

During slicing it reports "There is an object with no extrusions in the first layer". Now while I think I understand what it means, it clearly isn't the case here. It even seem to slice correctly but I cannot export it.

[image]

### Project file & How to reproduce

Just slice it.

[bug.3mf.zip](https://github.com/user-attachments/files/18341817/bug.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

macOS 15.2

### Printer model

Bear MK3S+

rtyr, 2025-01-08:
The notification is correct, since there is indeed an object with no extrusions. It is visible before slicing and also in object list.
[image]

closed 2025-01-08 by rtyr

tcurdt, 2025-01-08:
@rtyr to confirm: it's the "speckle" that is causing this?

Since everything else sliced without problems - why isn't this just a warning? instead of blocking the export?

Since the slicer has some fixing builtin - why can't it fix this (by removing it)?