prusa3d/PrusaSlicer #13930: Slicer crashes during gcode generation

rcmaniac25, 2025-01-08:
### Description of the bug

I loaded a multi-color painted model (https://www.myminifactory.com/object/3d-print-bamboo-dragon-cinderwing3d-x-bambulab-354514) and after setting up with the desired print settings, waited for it to slice. During the generating g-code stage, the slicer simply closed.

### Project file & How to reproduce

1. Get the files from https://www.myminifactory.com/object/3d-print-bamboo-dragon-cinderwing3d-x-bambulab-354514
2. Load in the "MMU_Prusa.3mf" version. If you try to open this as a project, it will simply load the model. So make sure to have the MMU printer selected or it will show up as single color.
3. Update every object to have wipe into infill
4. Let it slice

Project (only plate 1 has the issue. Plate 2 does not have the issue)
[Bamboo Dragon-Upload.zip](https://github.com/user-attachments/files/18343636/Bamboo.Dragon-Upload.zip)

Screen visual
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows 10

### Printer model

Prusa MK4S+MMU3

murk-sy, 2025-01-09:
If I had to guess it sounds like you have an 13th/14th gen Intel processor.

The reason it didn't crash once is (I believe) the fact that it went from a highly multithreaded to highly singlethreaded workload very quickly. If you sliced while running something else CPU intensive (like Prime95 stress test or something) I believe you should be able to slice reliably.

Check these:
- https://community.intel.com/t5/Blogs/Tech-Innovation/Client/Intel-Core-13th-and-14th-Gen-Desktop-Instability-Root-Cause/post/1633239
- https://www.intel.com/content/www/us/en/download/15951/intel-processor-diagnostic-tool.html

Duplicate issues:
- https://github.com/prusa3d/PrusaSlicer/issues/13483
- https://github.com/prusa3d/PrusaSlicer/issues/12387
- https://github.com/prusa3d/PrusaSlicer/issues/12236
- https://github.com/prusa3d/PrusaSlicer/issues/12671

rcmaniac25, 2025-01-09:
> If I had to guess it sounds like you have an 13th/14th gen Intel processor.

Nope. 6th Gen Intel

murk-sy, 2025-01-09:
The one time I predict it's a 14th gen thing without explicitly testing it ends up being not the problem... bah humbug

Reproducible reliably, I'll try to see if I can minimise the issue to fewer parts.

rcmaniac25, 2025-01-09:
I have tried combinations of the following

- Unfocusing application (click away but still visible)
- Backgrounding application (hide behind another window)
- leaving the models selected (when a 3mf is first loaded, all the models are selected)
- unselecting the models
- plater view
- slice view
- changing to the other bed before changing back (has the fun of not always showing that it's slicing, or sometimes just not slicing unless slice view is opened)

The one time it didn't crash it was background + plater view + unselected model + never clicked away from first bed. Same arrangement has not resulted in anything other then crashes.

I'm more wondering if it has to do with wipe into infill, as I know that is only processed during gcode generation phase and when I first made the project, it sliced without issue. But "click on 1-2 things and let slice" is a lot less mental occupation then "uncheck every wipe into infill" option.

Until PS maintains multi-color painting when doing cut/merge/split, I can't do a "merge" of all the individual objects without loosing the multi-color painting, otherwise I would merge everything into one object and set the wipe into infill (like the 2nd bed).

rcmaniac25, 2025-01-09:
I've been having issues compiling the project for some reason and it's probably a me thing, else I would make a ticket... but last time I managed to compile it, the debug version would give you the error without closing the application. But with release builds would simply close the application. So you're probably hitting a/the issue, but just with the debug build.

Jan-Soustruznik, 2025-01-09:
Hello, @rcmaniac25,  thank you for your report. I was able to reproduce your issue. I created an internal issue, no. SPE-2640.

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena

rcmaniac25, 2025-02-20:
Well it didn't crash, so good for me.