prusa3d/PrusaSlicer #13933: High resource (GPU) usage and flicker on macOS

meermanr, 2025-01-08:
### Description of the bug

When the PrusaSlicer window is just the right dimensions, the UI flickers as fast as possible and consumes ~91% GPU on my MacBook Pro (Apple Silicon, M2 Max), which increases total system power consumption 4x: 14 watts -> 54 watts.

This happened to me when I undocked my laptop from some external screens, and the window was reseized automatically without user input. I didn't use PrusaSlicer for some hours while I was doing other things, and didn't notice it under the other windows. I did however notice that my laptop's fans started, and then steadily ramped up enough to be heard. (Not normal for Apple Silicon!)

### Project file & How to reproduce

1. Create a project and populate mulitple plates (mine has three).
2. Go to the sliced view
3. Resize the window so the new plate indicator (top right) overlaps slightly with the legend (top left)
4. Observe that the UI flickers, and GPU raises and remains at ~90%.

3MF project file: [Garden Lawn - Grippy mat.3mf.zip](https://github.com/user-attachments/files/18351247/Garden.Lawn.-.Grippy.mat.3mf.zip)

Video of the issue: https://github.com/user-attachments/assets/57b9da56-8a13-40f1-8a7f-d8ade38ea4ff

h3. System info

Copied from Help > System Information

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 15.2 (Build 24C101)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   4.1.0 (4.1 Metal - 89.3)
Profile:      Core
Vendor:       Apple
Renderer:     Apple M2 Max
GLSL version: 4.10.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

macOS Sequoia 15.2 (Apple M2 Max)

### Printer model

Prusa XIS 5T

barneym, 2025-02-20:
I'm encountering this same issue.  It seems to happen with the default window size that Mac opens to, so particularly annoying.  Though confirm that playing around with window sizing seems to mitigate the issue.

doodlebro, 2025-03-11:
Prusaslicer always makes my LG DualUp and Dell monitors flicker uncontrollably and start burning in. As far as I understand the issue, Apple Silicon Macs are sensitive to certain grays on external monitors. It seems like prusa slicer just uses the exact gray values to cause issues.

closed 2026-09-29 by prusaslicer-bot[bot]