prusa3d/PrusaSlicer #13935: Printer continues to show offline in PrusaSlcer

greymattersf64, 2025-01-08:
### Description of the bug

I have been on vacation for 1 month and come back and before using my printer again was prompted to update the printer firmware and PrusaSlicer to the latest versions, which I did and painfully regret...

Since then the printer will NOT stay online. If I reboot the printer it will show as online for a few minutes, I can send a job to it and it will start printing and 15 minutes later when I go to check on the print progress through PrusaConnect it shows the printer as being off-line despite the fact it's still connected to the network and it's still printing. 

Multiple jobs remain in the queue despite manually deleting them, or that they are actually being actively printed. 

Overall responsiveness is far worse than using the version of PrusaSlicer and firmware that the device shipped with a few months ago. There seems to be some serious regressions introduced and I would very much like to have the stability and reliability that I had before these updates back ASAP. 
[Rotating head holder-5.3mf.zip](https://github.com/user-attachments/files/18352098/Rotating.head.holder-5.3mf.zip)

[image]

### Project file & How to reproduce

[Rotating head holder-5.3mf.zip](https://github.com/user-attachments/files/18352104/Rotating.head.holder-5.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Mac OS 14.2.1 (23C71)

### Printer model

MK4S

Jan-Soustruznik, 2025-01-09:
Hello, @greymattersf64 , Thank you for your feedback. We reviewed the issue you reported and concluded that the problem lies with PrusaConnect, not in PrusaSlicer. We’ve created an internal issue (CONN-706) for the Connect team.

Thank you, and have a great day!

greymattersf64, 2025-01-09:
Thanks for the response. Is there a way for me to roll this back to a previous version as it completely does not work and I have resorted to manually putting files on the USB drive and walking back and forth to the printer in order to print which is not ideal. 

Thanks!

Jamie

yba2cuo3, 2025-04-02:
As of today, April 1, 2025, the problem still persists.  Any updates from anyone?

closed 2026-09-29 by prusaslicer-bot[bot]