prusa3d/PrusaSlicer #13938: Arrange not working properly

TomBrodsky, 2025-01-09:
### Description of the bug

Auto arrange doesn't work properly with large parts. It force one item in center of the bed instead of group centered, resulting in ineffective printing workflow. For this particular part which needs to be printed 14 times it makes quite a difference.

Automatically arranged bed - it allows only one part per bed
[image]

Manually positioned parts
[image]

### Project file & How to reproduce

[VD0037334.zip](https://github.com/user-attachments/files/18358092/VD0037334.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Win 10+11

### Printer model

Trilab DeltiQ 2 Plus

RYUcze, 2025-04-07:
Can confirm this bahaviour, auto arrange should group models and align entire group to center, but it somehow forces first instance to center first, then tries to arrange secon instance.

TomBrodsky, 2026-09-22:
This still doesnt work in 2.9.6