prusa3d/PrusaSlicer #13941: Wipe movement at the end of each layer ignored

Split7fire, 2025-01-10:
### Description of the bug

My printer (hugely modernized ) has now Bowden extruder and I have to enable wipe. So wipe is working well, except last wipe in each layer. Firstly, I observed this on my machine because of strings at layer change. Later, when debugging GCODE it seems that printer just skipped part containing wipe movement. It is part from provided gcode:

```
.......
G1 X84.57 Y81.344 E24.99596
G1 X83.991 Y81.344 E25.01556
G1 X83.348 Y81.988 E25.04636
;LAYER_CHANGE
;Z:2.2
;HEIGHT:0.2
;TIMELAPSE_TAKE_FRAME
;WIPE_START
G1 F1440
G1 X83.991 Y81.344 E24.32591
G1 X84.57 Y81.344 E23.86753
G1 X83.531 Y82.384 E22.70372
G1 X83.531 Y82.962 E22.24614
G1 X84.602 Y81.89 E21.04636
;WIPE_END
G92 E0
G1 Z2.2 F180
SET_PRINT_STATS_INFO CURRENT_LAYER=11
G1 X102.186 Y133.083 F1800
G1 Z2.2 F180
G1 E4 F1200
;TYPE:Perimeter
;WIDTH:0.449999
G1 F900
G1 X101.551 Y133.02 E4.0216
.........
```
My guess that `;LAYER_CHANGE` triggers layer change routines.
Is this expected behavior?

### Project file & How to reproduce

[Gcode.zip](https://github.com/user-attachments/files/18372520/Gcode.zip)
[Project.zip](https://github.com/user-attachments/files/18372524/Project.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Версия 2.9.0

### Operating system

Bluefin (Fedora Silverblue 41)

### Printer model

FlyingBear P902 + Mainsail

3Delights, 2025-01-10:
I don't know how much you know about G-Code so:

The lines beginning with a semi-colon (;) are just comments and are ignored by the printer.  So as far as the printer is concerned that code snippet actually looks like this:

```
.........
G1 X84.57 Y81.344 E24.99596
G1 X83.991 Y81.344 E25.01556
G1 X83.348 Y81.988 E25.04636
G1 F1440
G1 X83.991 Y81.344 E24.32591
G1 X84.57 Y81.344 E23.86753
G1 X83.531 Y82.384 E22.70372
G1 X83.531 Y82.962 E22.24614
G1 X84.602 Y81.89 E21.04636
G92 E0
G1 Z2.2 F180
SET_PRINT_STATS_INFO CURRENT_LAYER=11
G1 X102.186 Y133.083 F1800
G1 Z2.2 F180
G1 E4 F1200
G1 F900
G1 X101.551 Y133.02 E4.0216
.........
```

The code performing the wipe is this bit:

```
G1 F1440
G1 X83.991 Y81.344 E24.32591
G1 X84.57 Y81.344 E23.86753
G1 X83.531 Y82.384 E22.70372
G1 X83.531 Y82.962 E22.24614
G1 X84.602 Y81.89 E21.04636
```

So this is the bit you need to debug.
G1 is a move command. It also can be used to specify movement parameters, in this case, Fnnnn means set the movement speed to 1440.

To move and print use G1 to move the nozzle tip top: Xnn.nnn Ynn.nnn tells it where to move to, and Enn.nnnn tell it how much filament to extrude as it goes.

Now the previous code block ended by moving the nozzle tip to X=101.551 and Y=133.02.
If I assume that the home/origin (0.0) is in the front left corner of the bed on your machine then the first line of the wipe code moves the nozzle to 83.911, 81.344 while extruding 24.32591 units of filament.  So in plain English it moves left and forward a little way while pushing out 24.3291s worth of filament. The rest follows the same pattern.  So the GCode translates as:
Set speed to 1440.
Move left and forward while extruding filament. (Diagonal)
Move right only while extruding filament. (Straight line to the right)
Move left and back while extruding filament. (Diagonal)
Move left and forward while extruding filament. (Diagonal)
Move right and forward while extruding filament. (Diagonal)

So now you need to go through your gcode and search for each wipe and see what it is doing and whether it is right or not.  You can find the start of each layer by searching for G1 Znn.nnn which moves the nozzle up by nn.nnn units.  Just above this will be the wipe code for the end of the last layer.

Hope this helps a bit at least...

Split7fire, 2025-01-11:
> The lines beginning with a semi-colon (;) are just comments and are ignored by the printer. 
I know that, but PrusaSlicer Gcode viewer don't display wipe movement either. It just jumps to to next layer. You can check that by yourself. Again, I don't know whether this expected behavior or not.

KoeSource, 2025-02-04:
Hello

PrusaSlicer-2.9.0

I have a similar problem.
See Images.

As soon as you print with retraction a WIPE block is inserted. 
The WIPE speed can only be limited with the rapid traverse (1).
I cannot limit the movements after the block with any setting(2).
This is very tedious, especially with the first layer.

[image]

[image]

[image]

closed 2025-12-03 by Split7fire