prusa3d/PrusaSlicer #13944: PrusaSlicer does not auto-renew its security token to PrusaConnect

scottdsanders, 2025-01-10:
### Description of the bug

I tend to leave PrusaSlicer running all of the time. Unfortunately the PrusaConnect tab, where I monitor my ongoing prints, doesn't seem to correct renew its security token so within a day, it expires and then I get a persistent flashing pop-up that says Error: Missing or invalid security token. My only recourse is to close and reopen the slicer (which usually makes me decide about saving/transferring profile info) or log out and in again which makes me go to my 2FA authenticator. Seems like if I leave the app open this should just work.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Version 14.6.1 (23G93)

### Printer model

Prusa MK4S (6.1.3)

TerPuc, 2025-01-13:
Dear @scottdsanders, thank you for the report. Before we will communicate with colleagues from Prusa Connect department, may I ask you to try something? Could you please try to reload the page? When the error occuers go to _Edit > Reload web content_. Please let me know if that fixes the error. Thank you!

scottdsanders, 2025-01-13:
That does appear to fix the issue. Thank you. I should have also pointed out in the bug report that I can still send new prints to the printer. Thanks.

-Scott

TerPuc, 2025-01-15:
Thank you for the additional information. I'm glad to hear that reload works! But this state still needs fixing. I have created issue in the internal system with number SPE-2644. And we will take a look at this with our colleagues from Prusa Connect department.

lukasmatena, 2025-02-18:
Some of the "logging out" issues are fixed [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1), hopefully all of them. Can you tell us if the problem disappeared for you in 2.9.1-alpha1? Thanks.

scottdsanders, 2025-02-22:
Thanks Lukas,

I’ve been using 2.9.1 alpha for a few days now and it seems like the issue is resolved. What I observe is that the PrusaConnect tab is refreshed almost every time I revisit it – which is ok. The authentication error is no longer a problem.

-Scott

lukasmatena, 2025-02-24:
Thanks for the update. Closing.

closed 2025-02-24 by lukasmatena