prusa3d/PrusaSlicer #13945: PrusaSlicer not running with flatpak when running on dedicated gpu

PredellaN, 2025-01-10:
### Description of the bug

When starting prusaslicer with "flatpak run com.prusa3d.PrusaSlicer", the application starts and works normally.
When starting with:
- "flatpak --env=__NV_PRIME_RENDER_OFFLOAD=1 --env=__GLX_VENDOR_LIBRARY_NAME=nvidia run com.prusa3d.PrusaSlicer"
- "switcherooctl -g 1 com.prusa3d.PrusaSlicer"
The application stop here:
`
nicolas@pop-os:~$ flatpak run --env=__NV_PRIME_RENDER_OFFLOAD=1 --env=__GLX_VENDOR_LIBRARY_NAME=nvidia com.prusa3d.PrusaSlicer 
--------------------------------------------------------------------------
Message: 21:35:42: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
nicolas@pop-os:~$
`
Starting the 2.8.0 AppImage version with switcherooctl -g 1 or with the environment variables specified before, the application starts on the dedicated GPU as intended. Other flatpak applications i am using run on the dedicated GPU without issue.
Enabling device=dri won't fix the issue.

### Project file & How to reproduce

- install the flatpak version of prusaslicer on a device with hybrid graphics
- run prusaslicer with either of the above mentioned commands

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Pop OS 22.04

### Printer model

any

Jan-Soustruznik, 2025-01-13:
Hello, @PredellaN , Could you check if PrusaSlicer has permission to use the GPU? You can add the necessary permissions using this command: `flatpak override --gpu=all com.prusa3d.PrusaSlicer`. Personally I would recommend [flatseal](https://flathub.org/apps/com.github.tchx84.Flatseal) to set/manage flatpak permissions in general.

PredellaN, 2025-01-13:
Hello @Jan-Soustruznik ,
```
nicolas@pop-os:~$ flatpak override --gpu=all com.prusa3d.PrusaSlicer
error: Unknown option --gpu=all
```
I assume you meant --device=all , however it doesn't solve the issue (it is already on by default).  --device=dri also doesn't solve the issue

benderl, 2025-01-27:
Same issue here with current Ubuntu 24.04 with slicer 2.9.0. I was previously using the AppImage without problems, but the flatpack refuses to start on th esecondary gpu.

iff7378, 2025-02-19:
Same issue Ubuntu 22.04. Appimage was working previously. Before I had the nvidia drivers installed on flatpak, prusa slicer would open but the 3d window wouldnt render. Instaled the Nvidia drivers and now it won't even run.

PredellaN, 2025-02-19:
workaround: there is an appimage build at https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous , functions correctly with discrete gpu

Samueru-sama, 2025-02-21:
> workaround: there is an appimage build at https://github.com/Samueru-sama/PrusaSlicer/releases/tag/continuous , functions correctly with discrete gpu

Hey [doesn't probonos fork work? ](https://github.com/probonopd/PrusaSlicer/releases/tag/continuous) it is more recent. 

See relevant issue: https://github.com/prusa3d/PrusaSlicer/issues/13653

My build is old and from this old PR https://github.com/probonopd/PrusaSlicer/pull/5

PredellaN, 2025-02-21:
> Same issue Ubuntu 22.04. Appimage was working previously. Before I had the nvidia drivers installed on flatpak, prusa slicer would open but the 3d window wouldnt render. Instaled the Nvidia drivers and now it won't even run.
> 
> Used flatseal to give permission to the GPU acceleration and wayland

ok, seems that the issue happens regardless of the session being X11 or wayland

SachCZ, 2025-03-04:
Hello this is most probably a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13618.

closed 2025-03-04 by SachCZ