prusa3d/PrusaSlicer #13949: Fan acceleration time

user32rtu, 2025-01-11:
When printing overhangs located in a row on several layers, the fan does not have time to gain momentum before the overhang begins. Sometimes the beginning of the overhang begins to bend. In SuperSliser I use the "Speedup time" function in the printer profile and it solves this problem. Some fans are quite inertial and take 2 seconds to spin them up.

Do something like this.
[image]

me25542, 2025-01-11:
This would be nice.

user32rtu, 2025-01-12:
> This would be nice.
Yes, there have already been many models that had to be sliced ​​in SuperSliser for this reason.

Tupson444, 2026-09-19:
still wanted

Jan-Soustruznik, 2026-10-21:
A similar feature request already exists. This request has been merge to keep topics organized, so please continue the conversation in discussion #15955.

closed 2026-10-21 by Jan-Soustruznik