prusa3d/PrusaSlicer #13951: PrusaSlicer for Windows requires WebView2 Runtime

JugglerLKR, 2025-01-11:
### Description of the bug

PrusaSlicer for Windows requires WebView2 Runtime, which is not always available.

### Project file & How to reproduce

Install clean version of Windows 10/11, install PrusaSlicer, click log in,

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11 x64

### Printer model

no matter

Jan-Soustruznik, 2025-01-13:
Hello, @JugglerLKR , It seems like a similar issue with a missing WebView2 DLL that was reported on Github. Could you try checking and installing it manually? You can find instructions here: https://github.com/prusa3d/PrusaSlicer/issues/13906#issuecomment-2580986962 Also, could you let us know if this resolves your problem? Thank you, have a nice day

JugglerLKR, 2025-01-13:
> Hello, @JugglerLKR , It seems like a similar issue with a missing WebView2 DLL that was reported on Github. Could you try checking and installing it manually? You can find instructions here: [#13906 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/13906#issuecomment-2580986962) Also, could you let us know if this resolves your problem? Thank you, have a nice day

Yes, WebView2 runtime resolved the problem. But the root cause is that not everyone wants to have anything "ms edge" in their systems.

Herrcookiekiller, 2025-01-17:
> Yes, WebView2 runtime resolved the problem. But the root cause is that not everyone wants to have anything "ms edge" in their systems.

I agree, installing MS Edge is like installing a virus. It adds so many background processes and things to autostart, and if you disable any of them, it stops working. I already figured out that they implemented the WebView Runtime in Prusa Slicer in version 2.8.1-rc1. Seems pretty unnecessary to me since everything already worked without it.

kocikdav, 2025-01-20:
Hi @JugglerLKR and @Herrcookiekiller,

Thanks for bringing this up. Microsoft Edge is a key part of how we display webpages for things like the Connect service, login, the Printables store, and the Physical Printer page. Without Edge, these features unfortunately won’t work.

That said, crashes due to missing libraries are definitely something we need to fix.

Thanks again for your understanding!

Avuja, 2025-02-16:
Fixed it for me by running the WebView2 Runtime offline installer.
Looks like they provide a shippable fixed version of the runtime so presumably just need to bundle in the .cab file with the slicer release.
This is on an ARM64 Win11 that did have Edge installed but I'd ran anti-telemetry tools including neutering Edge browser.

CRCinAU, 2025-02-18:
> That said, crashes due to missing libraries are definitely something we need to fix.

I mean, the correct fix would to remove the embedded web browser completely. It serves no real purpose other than self-promotion of the Prusa ecosystem.

Prusa has built their name and reputation on being open and focusing on quality products and software. Now it seems like its on the road to enshittification.

Whatever internal sales-droid lead meeting prompted this cause of action should be reversed.

MatyEjsr, 2025-05-21:
It shouldnt be a hard reqirement, i mean it on works on linux withou edge webview right?

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena