prusa3d/PrusaSlicer #13952: PrusaSlicer 2.9.0 Automatic painting crash with multiple beds

dreddpenguin, 2025-01-12:
### Description of the bug

I have not seen this reported elsewhere so I wanted to report what I ran into and traced down.
With the new multiple beds feature in PrusaSlicer 2.9.0, if you select a part on another bed and attempt to apply automatic paint for supports, the program will crash without any errors.

This has happened on every project file I have tested with multiple beds in use. I will attach one such project file.

If you first select the bed that part is on and then select the part and apply paints, there is no issue.

This screenshot shows the conditions for the crash, bed 1 is active but part on bed 2 is selected for painting on supports. 
Clicking Automatic painting will trigger the crash.
[image]

### Project file & How to reproduce

[Stargate - Earth Fleet.zip](https://github.com/user-attachments/files/18387885/Stargate.-.Earth.Fleet.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11

### Printer model

MK4

Jan-Soustruznik, 2025-01-13:
Hello, @dreddpenguin, Thank you for your report. I create a internal issue(SPE-2642). We will try to fix in some of new release. Have a nice day

guillermytho, 2025-01-25:
The crash occurs if you paint supports on a model from a bed that is not selected. Automatically selecting the bed where the object is would probably fix this.
If you first change beds and then create supports it won't crash.
BTW, I'm using the latest flathub linux version.

Jan-Soustruznik, 2025-02-18:
Closed as solved by the new release of [2.9.1-Alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1)

closed 2025-02-18 by Jan-Soustruznik