prusa3d/PrusaSlicer #13953: Slicer incorrectly displays sliced ​​layers in the bed description area

kabelkao, 2025-01-12:
### Description of the bug

When I slice an object and it overlaps the number that describes the current bed, it does not appear in the 1st layer in place of the layer number. But everything is fine in the gcode.
[image]
[image]

### Project file & How to reproduce

Place the sliced ​​object over the number that describes the bed.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Win 11

### Printer model

MK4S

neophyl, 2025-01-12:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13814  please read the comment in that thread.  Its a visual glitch and does not effect printing.

Jan-Soustruznik, 2025-01-13:
Closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/13814