prusa3d/PrusaSlicer #13956: Side panel customize with user selectable 'frequently changed settings

dsjove, 2025-01-12:
I like having the supports and infill settings right on the plater page side panel. There are a couple more settings that I would like to have there; 'support style' to name one. I would to have the ability to add user selectable 'frequently changed settings' to the side panel.

neophyl, 2025-01-12:
Please search for existing issues before opening new ones. This is a duplicate of a recent one https://github.com/prusa3d/PrusaSlicer/issues/13865 which is itself a duplicate of an older one. 
Opening duplicates doesn’t get things done quicker, it just makes more admin work for the devs.

dsjove, 2025-01-12:
Please add that request with suggestions on how do it most effectively to the main issue page and the new bug/feature pages. I would not have known to do that, aside from your comment.

neophyl, 2025-01-12:
It is on the main issues page. There are over 4 thousand open issues currently and growing.  The instructions when opening a new issue ask you to search and not create duplicates. Given how many new issues you opened in such a short space of time it appears you made no attempt to search for existing ones at all.

dsjove, 2025-01-12:
No. It is not. Screenshots attached. I get it. Issues are growing. That's what happens when organizations grow and products are rushed.

Where are these instructions? They are not on the pages I dealt with.

[image]
[image]

neophyl, 2025-01-12:
Sorry, my error, yes they should add the instructions from the Bug Report option to also cover the Feature Request button too.
[image]

More than one of your requests are covered under previous issues.

dsjove, 2025-01-12:
Yeah. Not surprised. It looks like somebody just added a duplicate of one of mine- that may already be a duplicate. 

The instructions - rather polite request - to search for previous needs to be in all three places; including the page that has the get started buttons. Because once the user starts, they are not likely not to stop. And without the polite request they aren't going to care nor know how annoyed you get.

We are just non-techie users making suggestions and specifying irksome things.

closed 2026-09-29 by prusaslicer-bot[bot]