prusa3d/PrusaSlicer #13972: Fuzzy skin out of position error when no walls or top layer is present

jtmx101, 2025-01-12:
### Description of the bug

Bug found while testing fuzzy skin settings.

While test printing fuzzy skin on the attached file, the last gcode position goes out of travel range.

I'm wondering if it's because my test part does not have top infill, and fuzzy skin is the last active setting before the print finishes.

### Project file & How to reproduce

Slice the supplied file. The last gcode position is attempting to move the tool head out of range in XY

This is a 350mm x 350mm voron. No custom scripts.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 11

### Printer model

Voron 2.4 350mm. .4mm nozzle

jtmx101, 2025-01-12:
[Uploading Fuzzy Skin Issue.zip…]()

Jan-Soustruznik, 2025-01-13:
Hello, @jtmx101 , could you try to attach the file again, please? I cannot dowload it. Thank you

closed 2026-09-29 by prusaslicer-bot[bot]