prusa3d/PrusaSlicer #13973: Failed to load Web Browser

killerkoffe, 2025-01-13:
### Description of the bug

Failed to load web browser. Unable to log in.
Note: Prusaslicer opens links in my web browser no problem.

[image]

### Project file & How to reproduce

1.) Configuration -> Configuration Wizard
2.) Log in
**OR:**
"log in" menu from GUI

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Microsoft Windows 10 Pro Version	10.0.19045 Build 19045

### Printer model

Original Prusa MK4s

killerkoffe, 2025-01-13:
Launched from the console, I see this log:
`[2025-01-12 17:25:57.820519] [0x00005bdc] [error]   Failed to create wxWebView object.`

killerkoffe, 2025-01-13:
I used Process explorer to get more info.

| Process          | CPU    | Private Bytes | Working Set | PID   | Description | Company Name   |
|------------------|--------|---------------|-------------|-------|-------------|----------------|
| prusa-slicer.exe | < 0.01 | 450,860 K     | 256,732 K   | 21440 | PrusaSlicer | Prusa Research |

**DLLs** <details>
| Name                     | Description                                                  | Company Name          | Path                                                                                                                         |
|--------------------------|--------------------------------------------------------------|-----------------------|------------------------------------------------------------------------------------------------------------------------------|
| advapi32.dll             | Advanced Windows 32 Base API                                 | Microsoft Corporation | C:\Windows\System32\advapi32.dll                                                                                             |
| bcrypt.dll               | Windows Cryptographic Primitives Library                     | Microsoft Corporation | C:\Windows\System32\bcrypt.dll                                                                                               |
| bcryptprimitives.dll     | Windows Cryptographic Primitives Library                     | Microsoft Corporation | C:\Windows\System32\bcryptprimitives.dll                                                                                     |
| cfgmgr32.dll             | Configuration Manager DLL                                    | Microsoft Corporation | C:\Windows\System32\cfgmgr32.dll                                                                                             |
| clbcatq.dll              | COM+ Configuration Catalog                                   | Microsoft Corporation | C:\Windows\System32\clbcatq.dll                                                                                              |
| combase.dll              | Microsoft COM for Windows                                    | Microsoft Corporation | C:\Windows\System32\combase.dll                                                                                              |
| comctl32.dll             | User Experience Controls Library                             | Microsoft Corporation | C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.19041.4355_none_60b8b9eb71f62e16\comctl32.dll |
| comdlg32.dll             | Common Dialogs DLL                                           | Microsoft Corporation | C:\Windows\System32\comdlg32.dll                                                                                             |
| CoreMessaging.dll        | Microsoft CoreMessaging Dll                                  | Microsoft Corporation | C:\Windows\System32\CoreMessaging.dll                                                                                        |
| CoreUIComponents.dll     | Microsoft Core UI Components Dll                             | Microsoft Corporation | C:\Windows\System32\CoreUIComponents.dll                                                                                     |
| crypt32.dll              | Crypto API32                                                 | Microsoft Corporation | C:\Windows\System32\crypt32.dll                                                                                              |
| crypt32.dll.mui          | Crypto API32                                                 | Microsoft Corporation | C:\Windows\System32\en-US\crypt32.dll.mui                                                                                    |
| cryptbase.dll            | Base cryptographic API DLL                                   | Microsoft Corporation | C:\Windows\System32\cryptbase.dll                                                                                            |
| cryptnet.dll             | Crypto Network Related API                                   | Microsoft Corporation | C:\Windows\System32\cryptnet.dll                                                                                             |
| cryptsp.dll              | Cryptographic Service Provider API                           | Microsoft Corporation | C:\Windows\System32\cryptsp.dll                                                                                              |
| d2d1.dll                 | Microsoft D2D Library                                        | Microsoft Corporation | C:\Windows\System32\d2d1.dll                                                                                                 |
| d3d11.dll                | Direct3D 11 Runtime                                          | Microsoft Corporation | C:\Windows\System32\d3d11.dll                                                                                                |
| DataExchange.dll         | Data exchange                                                | Microsoft Corporation | C:\Windows\System32\DataExchange.dll                                                                                         |
| dcomp.dll                | Microsoft DirectComposition Library                          | Microsoft Corporation | C:\Windows\System32\dcomp.dll                                                                                                |
[... cut ...]
| wtsapi32.dll             | Windows Remote Desktop Session Host Server SDK APIs          | Microsoft Corporation | C:\Windows\System32\wtsapi32.dll                                                                                             |
</details>

The following DLL is missing:
**C:\Program Files\Prusa3D\PrusaSlicer\WebView2Loader.dll**

This appears to be the problem. I have verified that the DLL exists, and is present on my computer, however it is not being loaded by prusa-slicer.exe.

I tried downloading 2.8.1, and I see the exact same issue.
WebView2Loader.dll is not being loaded by prusa-slicer.exe.

killerkoffe, 2025-01-13:
I see where **WebView2Loader.dll** is copied, but I don't see anywhere **WebView2Loader.dll** is actually linked. 

Can you add something like this:
```
add_library(WebView2Loader SHARED IMPORTED)
set_property(TARGET WebView2Loader PROPERTY IMPORTED_LOCATION "${CMAKE_CURRENT_BINARY_DIR}/${CONF}/WebView2Loader.dll")
target_link_libraries(PrusaSlicer WebView2Loader)
```
to here please:
https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/src/CMakeLists.txt#L204-L209

Thanks

TerPuc, 2025-01-13:
Dear @killerkoffe, thank you for you report. We will take a look at that. 

Internal issue created: SPE 2643

BlinkSTOP, 2025-01-25:
also reported similar on #14016

kocikdav, 2025-02-10:
Hello @killerkoffe thanks for your suggestions. 

We will check if the cmake regarding WebView2Loader.dll needs to be rewritten.

Meanwhile please check on your PC if Edge browser is installed. This issue might be similar as https://github.com/prusa3d/PrusaSlicer/issues/13906#issuecomment-2580986962. Also @BlinkSTOP I think missing Edge is very probably what is happening on your PC.

MatyEjsr, 2025-02-11:
edge shouldnt be a requirement - its microsoft bloatware, reroute through firefox or chromium embeded

Yigte, 2025-05-21:
This is how I solved the issue on my system.

https://github.com/prusa3d/PrusaSlicer/issues/13906#issuecomment-2898833324

closed 2026-09-29 by prusaslicer-bot[bot]