prusa3d/PrusaSlicer #13975: Adjust/assign the flow rate for each feature

SebKoAIM, 2025-01-13:
I am currently printing with a granule printer that requires an adjusted extrusion multiplier (EM) depending on the speed. Since different features are printed at different speeds, I am currently using the search and replace function in post-processing to change the override. Is there a possibility to introduce a feature like in Cura that allows setting the flow for individual features?

Thank you very much and best regards.

SebKoAIM, 2025-01-22:
Theoretically, this could also be done using different tools that all reference T0. Then, each tool could be assigned a different filament and thus a different EM. However, the print order is optimized for tool changes, so perimeter printing would not always occur first. A button to ensure the correct print order would therefore also be an option.

Thank you!

mstetson, 2025-02-06:
I believe this is a duplicate of #7200.

closed 2026-09-29 by prusaslicer-bot[bot]