prusa3d/PrusaSlicer #13978: Travel optimization is not working within a part

ronperek, 2025-01-13:
### Description of the bug

Travel optimization is not working inside a part. This increases time wasted on travel and stringing.

What travel looks like in Prusa single layer:
[image]

What travel looks like in Prusa for a whole part:
[image]

What I expect travel to look like on single layer:
[image]

What I expect travel to look like for the whole part:
[image]

### Project file & How to reproduce

[print-order.3mf.zip](https://github.com/user-attachments/files/18400278/print-order.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Linux

### Printer model

Prusa MK3S

beartronix, 2025-02-13:
Bump!
I have the same issue with a similar Design. Travel Moves take 24% of the time in my example, which could easily be reduced to 10-15% (my estimate...) and save hours of printing time, and thus, machine life! 

https://github.com/user-attachments/assets/2a6f4a34-5a8c-411a-943a-049a8a2f3df6
[image]

ronperek, 2025-05-30:
Fixed in 2.9.2. Thank you.