prusa3d/PrusaSlicer #13979: Enable Different Types of Support on the Same Print Object.

CodeGeaz, 2025-01-13:
Currently, the slicer generates uniform support structures across an entire print object. I propose adding a feature that allows users to specify and generate different types of support structures (e.g., tree-like support, linear, snug support, custom support angles) for different parts of the same 3D print object. This would offer more control over the support generation, potentially improving print quality, reducing material usage, and simplifying post-processing.

Use Case:
Complex Models: For models with varied geometries, some areas might benefit from tree support for better aesthetics and easier removal, while others might require more robust linear supports for stability.
Material Saving: Users could apply less dense or different support structures in areas where less support is needed, saving material and reducing print time.
Customization: Allow users to define support types based on specific geometric thresholds or areas manually selected within the slicer interface.

Suggested Implementation:
Introduce a new panel or section in the slicer settings for "Support Customization":
Support Type per Area: Allow users to define areas on the model where different support types are applied. This could be done via:
Manual painting of areas on the model for different support styles.
Setting conditions based on overhang angles or specific geometries.
Support Options: Include options like:
Tree Support: For areas where supports are hard to remove or where aesthetics are crucial.
Linear, Snug Support: For areas needing stronger support or where tree support might not be effective.
Custom Angles: Users could manually adjust the angle of supports for specific overhangs.

fftkatarn, 2025-07-12:
I totally need this! I'm currently printing a complex object for which I would need some grid supports for a large horizontal area and some organic supports for the rest.
It would be so useful in so many cases I really don't understand why it's not a thing yet!
Will this feature ever be implemented?

Tupson444, 2026-09-19:
still wanted

nlgordon, 2026-11-05:
I'm also going to hop in on this one. Large flat areas(grid) and small separate bits(organic) would really benefit from different types of supports.