prusa3d/PrusaSlicer #13981: "Aplication" typo in GLGizmoEmboss.cpp L3408

rasssta, 2025-01-14:
### Description of the bug

In PrusaSlicer/src/slic3r/GUI/Gizmos/GLGizmoEmboss.cpp there's a typo on line 3408 where it says "Aplication" instead of "Application":

https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/src/slic3r/GUI/Gizmos/GLGizmoEmboss.cpp#L3408

### Project file & How to reproduce

Look at:

https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/src/slic3r/GUI/Gizmos/GLGizmoEmboss.cpp#L3408

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

macOS

### Printer model

MK4S

rasssta, 2025-01-14:
I think that there should also be a space between 'one' and '(\"%2%\")'.

Overall, the grammar in lines 3408 and 3409 could be improved, such as:

"The application selected a similar one (\"%2%\"). You must specify a font to enable text editing."

Jan-Soustruznik, 2025-01-15:
Hello, @rasssta , thank you for your feedbak. We try to fix in some of next release. Have a nice day

closed 2026-09-29 by prusaslicer-bot[bot]