prusa3d/PrusaSlicer #13983: [BUG] Print bed is not probed for wipe tower, resulting in crashed nozzle and damaged print sheet

RichiH, 2025-01-14:
### Printer model

XL 5T

### Firmware version

6.1.3+7898

### Upgrades and modifications

1T->5T

### Printing from...

USB drive

### Describe the bug

The XL 5T is only probing the main print area, not the wipe tower area. The nozzle crashed into my print sheet where the wipe tower is supposed to be. As enabling input shaping still forces disabling of crash detection, the crashed nozzle scraped and damaged the print bed.

The XL then stopped because it detected "stuck filament".

This would be funnier if it hadn't irreparably damaged my print bed.

[image]
[image]
[image]

### Files

[bgcode.zip](https://github.com/user-attachments/files/18409904/bgcode.zip)

RichiH, 2025-01-14:
Happening on PrusaSlicer 2.9.0, Linux with flatpack

rtyr, 2025-01-15:
I was not able to reproduce it with provided 3MF on 2.9.0 win64.

Print area was correct in my case:
`M555 X25.4049 Y25.4088 W217.795 H127.365`

But print area is indeed incorrect in provided bgcode:
`M555 X25.4049 Y25.4088 W34.8217 H34.8307`

We will try to figure out how it could happen.

hejllukas, 2025-01-15:
Hi, we were able to reproduce your issue.
I created an internal issue, no. SPE-2645.

u89djt, 2025-01-15:
@hejllukas Is there a way of recognizing the risk during development of a project while you work on the bug?

lukasmatena, 2025-02-18:
Should be fixed in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). Closing.

closed 2025-02-18 by lukasmatena