prusa3d/PrusaSlicer #13984: Jagged / incoherent layers on specific model heights

Loosetooth, 2025-01-14:
### Description of the bug

When slicing models with specific heights (more on that below) some layers are jagged / incoherent.

Here is a screenshot of a problematic model. The top layer is split into many rough 'chunks':
[image]

### Project file & How to reproduce

The issue depends on the layer height settings.

The issue seems to be something like this:

When using Prusa Slicer, if the total height of the model minus the first layer height divided by the regular layer height results in a value that is ***exactly half a layer***, gaps may appear in the top layer.

This can be expressed mathematically as:

$$
\text{If } \frac{H - h_f}{h_r} = n + 0.5 \text{ for some integer } n, \text{ then gaps appear in the top layer.}
$$

Where:
- $H$ is the total height of the model,
- $h_f$ is the first layer height,
- $h_r$ is the regular layer height.

Whenever this condition holds true for any face of the model, issues **can** appear. Also when it is not the top layer.

In the case of the example project below, the first and regular layer heights are set to 0.2 mm.
The models are exactly 15.5 mm tall, as such the above 'half a layer' condition holds true.

The issue does not appear when the model has any other height. For example, even when adding 0.001 mm to the total model height will make the issue disappear completely.

**Project file:**
[Circlular-cutter-cookiecad.zip](https://github.com/user-attachments/files/18410291/Circlular-cutter-cookiecad.zip)

**STL files** included in the project:
[Circle-cookiecad-stls.zip](https://github.com/user-attachments/files/18410337/Circle-cookiecad-stls.zip)
One STL file is the original. The second STL file is the same model, but passed through a [mesh repair tool](https://www.formware.co/onlinestlrepair), just to be sure.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.8.0-linux

### Operating system

Nixos 24.11

### Printer model

Creality CR-10S

u89djt, 2025-01-14:
(From a fellow user)
An extrusion is only made where sufficient of the model occupies its volume. That judgement call is made based on local dimensions of the triangulated mesh. The lip of the mesh is thin, and the triangulation is coarse and uneven.
[image]
[image]
If you switch to the classic perimeter generator, which imputes extrusion direction diffently, you can see that the information in the mesh at the top doesn't obviously describe a continuous hoop way down at the scale of what an extrusion sees.
[image]
You need to give it something the slicer can interpret the way you want, and in this case, that happens when you tell it there's enough volume occupied in the top layer by bumping up the height. 
Bear in mind that wherever the model tops out, the physical print will top out at the top of a whole layer. When for example your layer height is 0.2 and your model height is 33.1, the physical print will be 3.2 high if any extrusions are generated in the last layer.

Loosetooth, 2025-01-15:
@u89djt I agree with your points that it is a thin wall (smaller than 0.4mm) and that it might be the reason that the perimiter generation acts weird here.

However, here is an example of a flat surface that is much wider than 0.4mm, but behaves in a similar incoherent way.
[image]
[image]

What would be the 'reason' for the behavior in this case? If the flat surface is slightly higher or lower in certain parts, that should be fine and not result in a partial path, no?
By the way, this only happens when the condition of exactly half a layer height holds true as explained above.

**Project file:**
[IMG 1273-cookiecad_fixed.zip](https://github.com/user-attachments/files/18422523/IMG.1273-cookiecad_fixed.zip)

**STL file:**
[IMG 1273-cookiecad_fixed_STL.zip](https://github.com/user-attachments/files/18422597/IMG.1273-cookiecad_fixed_STL.zip)

u89djt, 2025-01-15:
The mesh wiggles up and down a bit.
[image]

u89djt, 2025-01-15:
As does the cookie cutter. Good call.
[image]

Loosetooth, 2025-01-15:
Nice find, thanks for that.

So I guess the issue could be summarized as follows:
If both conditions hold true:
1. the average height of the surface of a model is at exactly half a layer-height. (see detailed explanation above)
2. the exact heights of the vertices that make up this surface vary a little (for example a couple of thousandths of a millimeter)

Then the path generation algorithm in Prusa Slicer will generate paths for the 'high parts' of the surface, but not for the 'low parts' of the surface.
Resulting in an inconsistent path for that layer.

u89djt, 2025-01-15:
2, yes, but the first point doesn't have to hold - if the surface meanders either side of the half way point in any way you'll still have an intermittent perimeter. There will be small model height changes up or down within the scatter of the mesh heights that would still exhibit the phenomenon.
I think we don't know whether thickness has any effect in the cookie cutter case, but it seems amply explained by height.

u89djt, 2025-01-15:
The perimeters you're seeing are the best possible match when extruding fixed-height layers of plastic to match the model you gave it.
For a model that doesn't stop at a layer boundary, the correct thing to do is to change the boundaries.
[image]
[methodCirclular-cutter-cookiecad.zip](https://github.com/user-attachments/files/18426195/methodCirclular-cutter-cookiecad.zip)
I don't know if there are any slicers that work differently. Non-planar extrusion is a whole other can of (variable thickness) worms that currently requires assertive hacking for special cases.

closed 2026-09-29 by prusaslicer-bot[bot]