prusa3d/PrusaSlicer #13987: [Prusa XL Toolchanger] Unchangable Dwell time setting  on toolhead change causes blobs on print

Stippy19, 2025-01-15:
### Description of the bug

After completing a layer in one color, the toolhead freezes for a little over a second before parking the head. In this time, the toolhead may* stay on the printed part, causing blobs by oozing filament. This seems to be caused by "G4 S1.500 ;" command in the CP TOOLCHANGE START block. However, it does not seem to be able to somehow change that by any settings in the slicer.

*Additionally to this setting not being changable: Using two toolheads, the first toolhead goes to the wipe tower, then does the 1.5s dwelling and then docks that toolhead while the second toolhead does the dwelling on the printed part and then is docked directly. I am not sure, if there is another option in the slicer I am missing for that.

Issue originally found by reddit user "finestaut", just matching my issue I had the same day. I am just writing his findings here, to be sure it's seen for Slicer development.
[Link](https://www.reddit.com/r/prusa3d/comments/1i1krd3/15_second_dwell_between_tool_changes/?share_id=EqY4lsEcdpw4sarFWZKAP&utm_content=1&utm_medium=android_app&utm_name=androidcss&utm_source=share&utm_term=1)

### Project file & How to reproduce

[case_no_toggle.zip](https://github.com/user-attachments/files/18430337/case_no_toggle.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Windows 10

### Printer model

Prusa XL

Mike-Flexicon, 2025-01-16:
I have fixed mostly fixed this with custom code in the "tool change G-code"   It drove me crazy that one head would go immediately to the wipe tower and the other would go to the part.. dwell.. then go to the wipe tower.  This code fixes that for me, besides having to update the location of the wipe tower when it's moved.

Printer: Modix Big-60
Firmware: RepRap
Board: Duet 2 wifi with duex
Physical Extruders: 2

My custom toolchange G-code:
{if next_extruder == 0}
G91
G0 Z5
G90
G0 X474 Y346 F4800  ; Go to Wipe tower. This should be a parameter instead of having to update it when I move the wipe tower
G91
G0 Z-5
G90
T0
{else}
T1
{endif}

pipcorona, 2025-02-14:
I am seconding this issue. I don't know when it changed, but older sliced files of mine did not have this issue. I recently re-sliced some files and noticed at the end of the print the tool head dwelled on the part before parking. As I was printing TPU, it was leaving a blob on the print. I looked into the gcode and found the same G4 S1.500; code. Manually changing that to S0 after slicing fixed the issue.

It would be nice to be able to control this option in the slicer, or have it where it only dwells if the nozzle is over the purge tower but not over the part.

kubispe1, 2025-02-17:
Hi everyone,

It looks like the 1.5-second pause G4 S1.5, originally intended for ramming in the MMU3, has unintentionally made its way into some tool changer G-code. This pause is meant to ensure the melt zone is warm enough before ramming, which helps remove 100% of the melted filament—but it doesn't make sense in a tool changer setup and should be removed.

This is a bug, and we will suppress the pause in an upcoming Final release. The pause should only appear before the wipe tower in an MMU3+MK4. 
Thanks for you help
SPE-2687

there is also X report, so I am linking it together: 
https://x.com/gonecozycrafts/status/1891140558220923109

Stippy19, 2025-02-18:
> Hi everyone,
>
> It looks like the 1.5-second pause G4 S1.5, originally intended for ramming in the MMU3, has unintentionally made its way into some tool changer G-code. This pause is meant to ensure [...]

Good to know you found  the bug :)
I don't think it's fixed in the current new version 2.9.1, right?

lukasmatena, 2025-02-18:
> I don't think it's fixed in the current new version 2.9.1, right?

It's not fixed in 2.9.1-alpha1 yet.

legomind, 2025-02-24:
While waiting for the fix, here is a post-processing script to remove the G4 commands. Since post processing scripts are not compatible with binary gcode, you must turn that off in printer settings. Forgive the uglyness, python is not my programming language of choice.

```python
#!/usr/bin/python
import sys
import re
import os

sourceFile=sys.argv[1]

# Read the ENTIRE g-code file into memory
with open(sourceFile, "r") as f:
    lines = f.readlines()

destFile = re.sub('\\.gcode$','',sourceFile)

search_block = False
with open(destFile, "w") as of:
    for lIndex in range(len(lines)):
        oline = lines[lIndex]
        # Parse gcode line
        parts = oline.split(';', 1)
        if len(parts) > 1:
            comment = parts[1].strip()
            
            if comment == "CP TOOLCHANGE UNLOAD":
                search_block = True
            
            if comment == "CP TOOLCHANGE END":
                search_block = False
        
        elif search_block:
            # Parse command
            command = parts[0].strip()

            if command == "G4 S1.500":
               continue 
            
        # Write original line       
        of.write(oline)
of.close()
f.close()
```

lukasmatena, 2025-02-27:
Fixed in [2.9.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-beta1). Closing.

closed 2025-02-27 by lukasmatena