prusa3d/PrusaSlicer #13994: czech language in PrusSlicer 2.9.0

jhquentin, 2025-01-16:
Hi,
I am running

PrusaSlicer
Version:   2.9.0
Build:     PrusaSlicer-2.9.0+flathub.org

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 6.12.9-zen1-1-zen x86_64
Total RAM size [MB]: 32,988MB
OpenGL installation
GL version:   4.6.0 (4.6 (Core Profile) Mesa 24.3.1 (git-c815d651b8))
Profile:      Core
Vendor:       Intel
Renderer:     Mesa Intel(R) Iris(R) Xe Graphics (RPL-P)
GLSL version: 4.60.0
Textures compression:       Enabled
Mutisampling: Enabled (4 samples)
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_gpu_shader_int64
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_gpu_shader_int64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_post_depth_coverage
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_ballot
GL_ARB_shader_bit_encoding
GL_ARB_shader_clock
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shader_viewport_layer_array
GL_ARB_shading_language_420pack
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map_array
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ATI_blend_equation_separate
GL_ATI_texture_float
GL_EXT_EGL_image_storage
GL_EXT_EGL_image_storage_compression
GL_EXT_EGL_sync
GL_EXT_abgr
GL_EXT_blend_equation_separate
GL_EXT_debug_label
GL_EXT_demote_to_helper_invocation
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_memory_object
GL_EXT_memory_object_fd
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_pixel_buffer_object
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_semaphore
GL_EXT_semaphore_fd
GL_EXT_shader_framebuffer_fetch
GL_EXT_shader_framebuffer_fetch_non_coherent
GL_EXT_shader_integer_mix
GL_EXT_shader_samples_identical
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shadow_lod
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_EXT_vertex_attrib_64bit
GL_IBM_multimode_draw_arrays
GL_INTEL_blackhole_render
GL_INTEL_conservative_rasterization
GL_INTEL_performance_query
GL_INTEL_shader_atomic_float_minmax
GL_INTEL_shader_integer_functions2
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_framebuffer_flip_y
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_const_bandwidth
GL_MESA_texture_signed_rgba
GL_NV_alpha_to_coverage_dither_control
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fragment_shader_interlock
GL_NV_packed_depth_stencil
GL_NV_texture_barrier
GL_OES_EGL_image
GL_S3_s3tc
</details>

on arch linux, my locale is:

localectl status
System Locale: LANG=en_US.UTF-8
               LC_TIME=C.UTF-8
    VC Keymap: cz-lat2
   X11 Layout: cz

locale -a
C
cs_CZ.utf8
C.utf8
en_GB.utf8
en_US.utf8
POSIX

When I wuold like to change my app language to Czech, it is not displayed in Configration - Language. Only English is presented.

I tried also

flatpak run --verbose --env=LANG=cs_CZ.UTF-8 --env=LC_ALL=cs_CZ.UTF-8 com.prusa3d.PrusaSlicer
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /home/quentin/.local/share/flatpak
F: Opening user flatpak installation at path /home/quentin/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Skipping parental controls check for app/com.prusa3d.PrusaSlicer/x86_64/stable since parental controls are disabled globally
F: Opening user flatpak installation at path /home/quentin/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: /var/lib/flatpak/runtime/org.gnome.Platform/x86_64/47/e075b6c52ef5c8424d5d1f649305a91f1db3b0c893ddef32515802616ee86178/files/lib32 does not exist
F: Allocated instance id 216991704
F: Add defaults in dir /com/prusa3d/PrusaSlicer/
F: Add locks in dir /com/prusa3d/PrusaSlicer/
F: Allowing homedir access
F: Allowing x11 access
F: Running 'bwrap --args 39 -- xdg-dbus-proxy --args=41'
F: Running 'bwrap --args 39 -- entrypoint'

But with no success too.

Can you help me?

Jan-Soustruznik, 2025-01-17:
Hello, Jan, 

When installing PrusaSlicer using Flatpak, only English and the system language are installed by default. In case you want to switch PrusaSlicer to another language, it is necessary to configure Flatpak to use these languages first. For example

`sudo flatpak config --set extra-languages "cs_CZ"`

will configure Czech for Flatpak. It is necessary to reinstall the application after.
Running the command with additional --user parameter will only configure the languages for current user, in which case sudo is not required.

source: https://github.com/prusa3d/PrusaSlicer/releases 

Issue closed as resolved. Thank you, have a nice day

closed 2025-01-17 by Jan-Soustruznik

jhquentin, 2025-01-17:
Hi
I did

***@***.*** ~ $ sudo flatpak config --set extra-languages "cs_CZ"
***@***.*** ~ $ flatpak run --verbose com.prusa3d.PrusaSlicer
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path 
/home/quentin/.local/share/flatpak
F: Opening user flatpak installation at path 
/home/quentin/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Skipping parental controls check for 
app/com.prusa3d.PrusaSlicer/x86_64/stable since parental controls are 
disabled globally
F: Opening user flatpak installation at path 
/home/quentin/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: 
/var/lib/flatpak/runtime/org.gnome.Platform/x86_64/47/e075b6c52ef5c8424d5d1f649305a91f1db3b0c893ddef32515802616ee86178/files/lib32 
does not exist
F: Cleaning up unused container id 2777601209
F: Allocated instance id 2007976059
F: Add defaults in dir /com/prusa3d/PrusaSlicer/
F: Add locks in dir /com/prusa3d/PrusaSlicer/
F: Allowing homedir access
F: Allowing x11 access
F: Running 'bwrap --args 39 -- xdg-dbus-proxy --args=41'
F: Running 'bwrap --args 39 -- entrypoint'
--------------------------------------------------------------------------
Message: 09:49:32: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------

but I am not still able to change language.

Jan Hladík
***@***.***
Mobil: +420 603 324 153

jhquentin, 2025-01-17:
> Hi @jhquentin <https://github.com/jhquentin>, it's necessary to
> reinstall the flatpak app after you set new language. Hope this helps!

Hi, after reinstall it is ok. Thank you. You close the ticket.

Jan Hladík
***@***.***
Mobil: +420 603 324 153