prusa3d/PrusaSlicer #13995: Nozzle crashing on the object, causing a layer shift, when using supports and adaptive layers height

guivo, 2025-01-16:
### Description of the bug

After the upgrade to 2.9.0 I am having trouble using variable layer height in a model were I need supports. In some areas of the model, but not always in the same spot, the nozzle travels from the support to the model, hitting the model. When the change in layer height is particularly small (e.g about  .1 mm or less) the nozzle pushes hard enough and the motor skips, causing a layer shift.

### Project file & How to reproduce

1. Load a model that require supports from the bed
2. Select 0.2 mm high layer or less
3. enable the variable layer size for the model, default parameters
4. slice the file

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Window 11

### Printer model

Creality CR6 SE

u89djt, 2025-01-17:
(Fellow user following the hub) I haven't had this problem, but I'd like to know if there's a risk.
When you get a collision, post the exact 3mf file that you used for that print and a screenshot of the location of the issue in the preview. Check the bug submission guidelines to see if I've missed anything.
You can see the travel moves in the slice preview by switching them on in the feature type view with the icon at the bottom of the pane.
You need to zip the 3mf file to post, but you can drag and drop or paste the screenshot - capture the whole window - showing the travel move of interest in the slice preview.

guivo, 2025-01-19:
I should have added the file as requested.

I have double checked the printer status several time, no mechanical problems are present. I had issue printing this model, when using adaptive layer high to try to save time, none while using a regular constant layer height.

The issue happened, as mentioned, several time, always within the first 5 mm. Looking more carefully, with the default options there are a few layers that have a quite small layer height, compatible with the areas in which the problem happened. 

[PrusaSlicer_issue.zip](https://github.com/user-attachments/files/18470137/PrusaSlicer_issue.zip)

u89djt, 2025-01-19:
(I'm a fellow user) I would guess that you're sailing close to the wind with those two 0.09mm layers, which are smaller than the stated performance of that printer's 0.1mm resolution https://www.creality.com/products/cr-6-se-3d-printer 
I believe absolute height accuracy is particularly important close to the bed, so it may be that you can't print exactly that on that printer. 
But, that's just based on the prima facie evidence. If it is the answer, the solution would just be to back off layer density a bit. But a bit more digging is warranted.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]