prusa3d/PrusaSlicer #13998: PrusaSlicer 2.9.0 multimaterial painted surface depth

collurpf, 2025-01-17:
When I print a multi-colored part that I used "multimaterial painting" to alter the color, it seems to waste quite a bit of the filament on the colored surface.  Is there a way to reduce this?  In the example attached, the yellow test was printed on the side of a cylindrical surface, but it prints very far into the part.  I would like to limit it to a few mm rather than the existing depth of roughly 125 mm.  This would, I'd think speed up the print as more of the interior would be red infill instead of the current yellow "text".

[image]

m29corey, 2025-01-23:
This should be the setting you're looking for:

- Print Settings
  - Multiple Extruders
    - Advanced
      - Maximum width of a segmented region

It defaults to 0 (auto) which maximizes adhesion between the colored portions.
If you have it too shallow, you risk the part coming off.

Good luck.

collurpf, 2025-01-23:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

Awesome, that did the trick.

Thank you mysterious expert person!

Regards,

Pete Collura
Senior Engineer-Mechanical
FANUC America Corp.

closed 2026-09-29 by prusaslicer-bot[bot]