prusa3d/PrusaSlicer #14001: REPO QUALITY OF LIFE - Please open Prusa-Development/PrusaSlicer-Actions

ddavidebor, 2025-01-18:
**Is your feature request related to a problem? Please describe.**
I would like to contribute to PrusaSlicer. However, while I can build and test locally, I cannot take advantage of GitHub Action to build a draft release in the same way Prusa does. 

Several github actions the repo relies on are currently closed and located in "Prusa-Development/PrusaSlicer-Actions"

**Describe the solution you'd like**
Please make the relevant actions public. If you have parts of the actions that you'd like to keep private, you can easily set it up so they only on your repo.

dunkmann00, 2025-02-10:
I agree that this would be useful. It would be nice to be able to point others to a GitHub Actions build of a branch that has a feature you want to share or test out. This could lower the bar for getting people to try new features created by the community, which might be good info for Prusa as well.

Also SuperSlicer, OrcaSlicer, Cura, and (somewhat ironically perhaps) BambuStudio all have the GitHub Actions necessary to build their slicer, available for all to use.

iammattcoleman, 2026-09-03:
Does Prusa-Development/PrusaSlicer-Actions contain the build script referred to in [these comments in `version.inc`](https://github.com/prusa3d/PrusaSlicer/blob/6f510128d7c2e543b62919b74bea7e876f564205/version.inc#L1-L2)? If that's the case, then, as I described in [this comment](https://github.com/prusa3d/PrusaSlicer/issues/6630#issuecomment-5520196560), submitting a fix for the errors in the macOS app bundle's metadata will break the build script. Opening Prusa-Development/PrusaSlicer-Actions would enable me to submit a complete fix for the issue without worrying about breaking any automation.