prusa3d/PrusaSlicer #14003: Custom Fonts not usable in Text tool on MacOS 12

rocket-lawn-chair, 2025-01-18:
I have installed a custom font in both the User and System folders, but it isn't available for use in the text tool in the font selector for embossing or raised text. PS2.9.0 was installed first, before the font.

Font was installed through FontBook and through Copy/Paste in all Font folders.

Jony01, 2025-01-21:
Is the installed font TrueType?

rocket-lawn-chair, 2025-01-23:
Yes it's a ttf file.

Font is usable normally in other programs.

Jony01, 2025-01-24:
Could you provide font file for check?

rocket-lawn-chair, 2025-02-17:
Sorry for the late reply.

I would prefer not to share the font as it is a custom font of my handwriting and includes my signature.

The strange thing is that when I first installed 2.9, the font was visible in the text tool. But the font was an older version, when I moved it to the User directory, it was no longer visible. Putting it back in the system directory did not make it visible again.

Is it possible that 2.9 only builds it's list of available fonts when it is installed instead of every time it is loaded?

Jony01, 2025-02-18:
App try to reload list of the installed fonts in the OS, every time one open Text tool window.
But alse keep list of Failed fonts(and skip them from OS list), because test that loading fail is not cheap.
In case you create invalid font and refresh list the font faceName become depricated.
Also there is an hash for loaded list of fonts(based on faceNames) to not process it so often.

To remove font cache delete file: `/PrusaSlicer/cache/fonts.cereal`
(where PrusaSlicer is folder with configuration you could achiev by `PrusaSlicer/Help/Show Configuration Folder` )
[image]

Be careful history of font evolving is strange and left there a lot of exceptions for name etc.

rocket-lawn-chair, 2025-02-19:
Thank you for your guidance. I may ask additional questions when I try this fix.

Is there a recommended location for Fonts on MacOS? I know there is the User vs System libraries.

Jony01, 2025-02-20:
chat GPT:
In macOS, fonts are stored in two primary locations: **User Library** and **System Library**. The key difference between these locations lies in the scope of access and usage.

1. **User Library (~/Library/Fonts)**:
   - **Location**: This is a folder within the user's home directory.
   - **Scope**: Fonts stored here are only available to the specific user account.
   - **Customization**: Users can freely add, remove, and manage fonts in this folder without affecting other users on the same system.
   - **Use Case**: Ideal for individual user-specific fonts that shouldn't be shared across all accounts on the Mac.

2. **System Library (/Library/Fonts)**:
   - **Location**: This is a system-wide folder that is outside of individual user accounts.
   - **Scope**: Fonts stored here are available to all users on the system.
   - **Permissions**: Requires administrative privileges to modify or install fonts here, making it less flexible for non-admin users.
   - **Use Case**: Best for fonts that should be accessible across all accounts on the Mac, such as system fonts or fonts used by applications.

### Key Differences:
- **Access**: User Library fonts are only for the user that installs them, whereas System Library fonts are available for all users.
- **Permissions**: System Library fonts are protected by system-level permissions and require admin rights to modify. User Library fonts can be modified without admin privileges.
- **Scope of Use**: User Library fonts are limited to the specific user’s profile, while fonts in the System Library are shared across the entire system.

In essence, **User Library** is meant for user-specific customization, while **System Library** is intended for system-wide font access.

rocket-lawn-chair, 2025-02-20:
Is there any difference between the User and System Libraries in how PS 2.9 access them, or are they the same? Is one recommended over the other for compatibility. I am the only user of the system.

Jony01, 2025-02-21:
List of installed fonts is made by [WxWidgets](https://github.com/wxwidgets/wxwidgets/releases)
In [PS 2.9.0](https://github.com/prusa3d/PrusaSlicer/blob/5c7888b11453eb96a95b49dfd4cbd3b56db64eb8/deps/%2BwxWidgets/wxWidgets.cmake#L31) is 
`323a465e577e03f330e2e6a4c78e564d125340cb` which is wxWidget commit --> version v3.2.0

ShawnPierce, 2026-09-23:
This is True for windows systems as well, I installed a Intel Mono Font downloaded from the official Intel GitRepo then then I went in and added Text no fonts showed up (the drop down was empty) following the suggestion from @Jony01 and removing the Fonts.Clear from the cache folder then closing and reopening, add Text, the Fonts including the ones I added for Intel Mono are available and show up. Looking back in the cache folder the fonts.cereal file is back. So it appears that if the system fonts change the Prusa Slicer is broken until manually clearing the cache file. This Issue was very handy and found searching.