prusa3d/PrusaSlicer #14012: [BUG] manually added "block Supports Areas" ignored after adding a Modifier to the model

funtomaas, 2025-01-19:
### Description of the bug

Manually Added areas for "no support" are ignored if I add a modifier (-Cube with 100% Infill) afterwards on the top of the 3d-model. 
Support-Type: Organic.

### Project file & How to reproduce

- selected areas example: 
[image]

- without modifier on top: 

[image]

- modifier: 

[image]

- after slicing: supports in "no support-areas":

[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

 Windows 11 Home;  Version	10.0.26100 Build 26100

### Printer model

Mini+

u89djt, 2025-01-19:
You need to zip the project file to post it.

funtomaas, 2025-01-23:
[R_P_HOOK_v0p6.zip](https://github.com/user-attachments/files/18521132/R_P_HOOK_v0p6.zip)

here a projeckt file for this issue. 

INFO: There is a second MAJOR Issue in this Project, with the actual slicer (I checked an older Slice to be shure it is no Firmware-Bug, so sorry for my delayed answer). 
I'll describe it here first, maybe we need to open a new issue for it(?):
the print makes a GAP in the height where the  overhang of the hook starts! 
I tried a lot of changes (Organic support, standard support, manual added additional support etc.)  here a Photo of the catastrophical behaviour:  (with standard Supports)

[image]

u89djt, 2025-01-23:
The mesh doesn't have any errors. It does have unnecessary and unintended features: much higher density on curves than can be reproduced in extrusions, and tiny jagged over hangs at the base. I can't find anything that seems to be causing the problem in a way we should expect. The overhangs seem to have been successfully painted in the example file, and they're obviously ok without the modifier shape.
Noteworthy: the problem doesn't happen with grid or snug supports.
I don't have any thoughts on the gap.
That's my bit of community triage so far. Let me know if you want me to explore more.

[image]

funtomaas, 2025-01-23:
hm, your reply is for my second bug (the Gap) - thanx but I have no idea... 

The Main issue above with the "supports where suppors shouldn't appear" is the same with Slicer Version 2.8.1+win64 -I have it tested now...

u89djt, 2025-01-23:
Can you post the project file or files that gave you the gap? Bug hunting needs all the details to be absolutely clear and definite for everyone looking at it.

neophyl, 2025-01-23:
Looking at the geo of your object in blender its 3dprint tool reports 216413 Zero faces.  

[image]

Basically a lot of those are going to be unpaintable.  If you have supports defined by angles it will create support for ANY face that meets the angle criteria EXCEPT those that are painted with a blocker.  Given there's a bunch of tiny tiny faces at all sorts of angles that can not be painted I suspect that's what is causing you the support around the bottom.

If you add a modifier of type Support blocker over the base it stops the support as it doesn't care, it effects EVERY face inside its area.  Just stick a box or slab over that area.

Its for reasons like this that setting the support settings to "For Support Enforcers only" and then Painting on where needs supports tends to be a better methodology.  Don't bother blocking, just define where you want them in the first place.

u89djt, 2025-01-23:
Good spot. I ran through meshlab's tools and must have missed zero faces. I didn't know blender had a print analysis tool. 
I guess the grid supports have a different filter for things they generate support for?

u89djt, 2025-01-23:
I've rebuilt the object in Fusion 360 (file included). This project has the original object front left and a copy of it front right.
My rebuild is at the back with painted blocker on the one on the left, and the right is a direct copy of it. 
There's more going on here. A second copy of the original object doesn't exhibit the problem. There's some support where OP missed painting the blocker. I think that says that the painting was successful?
But my shiny new object exhibits the problem in the original and a copy. I wonder if the precise shape of the infill modifier is important.
So we have different behaviour in two copies of the same entity, but the same behaviour in two copies of something intended to be the same effective entity.
BTW, Meshlab doesn't think there are any zero triangles. Does the blender extension have a threshhold setting for calling triangles zero?

[RemasterAndOGTwice.zip](https://github.com/user-attachments/files/18526553/RemasterAndOGTwice.zip)

[image]

funtomaas, 2025-01-24:
> Can you post the project file or files that gave you the gap? Bug hunting needs all the details to be absolutely clear and definite for everyone looking at it.

as mentioned above: it is the same Projekt file :)

u89djt, 2025-01-24:
That project file slices to the photo in your first screenshots. The gap is produced in a different project file with grid supports. This is github: detailed situational awareness is essential. Something is happening that you don't understand, so you mustn't gloss over details. The only way for you to be absolutely sure that someone who can understand it has the correct information is - as explained in the submission guidelines - the project file just before you hit slice.

funtomaas, 2025-01-24:
Aha, sorry for that, I didn't realize it so far, but: you can ignore the Photo,I could send one with other supports too but I have removed the Support. As I notedbefore: I tried lots of Support-configs: the Effect with the gap is the Same. (So the Foto is just an example)

u89djt, 2025-01-24:
A lot can change between releases of the software and profile versions. I'll check in tomorrow :)

funtomaas, 2025-01-26:
@neophyl (for the Support issue)...
> Its for reasons like this that setting the support settings to "For Support Enforcers only" and then Painting on where needs supports tends to be a better methodology. Don't bother blocking, just define where you want them in the first place.

Thx. I KNOW how to handle this error behaviour, but I guess the development Team of the prusaslicer should know that there is this misbehaviour. In my eyes, even if there are many zero-faces or other modelspecific critics: if I mark the "no-Support-Areas" it works perfectly as long as I don't add a modifier AND change the infill. (I didn't try other modifications yet). That is the reason why I would say it IS a bug that should be fixed sometimes... 
so I'll be happy to wait for the fix (and use workarounds meanwhile.)
:)

funtomaas, 2025-01-26:
@Jan-Soustruznik are you admin? could I separate the parts of this issue that address the "Gap" at the overhang towards a new issue anyhow? .. an could you tell me how? or if not: could you do so?  because we have two different issues discussed here and I guess they should better be handled (and solved) separately... (sorry for that)

closed 2025-10-23 by Jan-Soustruznik