prusa3d/PrusaSlicer #14013: Ignoring Extruder-Selection for Bridge-Infill

AshmodaiVirolac, 2025-01-19:
### Description of the bug

For a Multimaterial-Print on XL-5-Tool  i have two little "Windows" in a Case-Blend that should be printed in clear PETG while the rest of the Object is printed in Black PETG.
For the Windows i use 100% infill and all layers get printed in clear PETG, beside one:
On the Layer where the Main-Body get the bridge-Layer, slicer ignores the Extruder-settings for the Windows and print that Layer in Black PETG too.

### Project file & How to reproduce

Loading a Multi-material-Object where one Object should be printed massively from selected Material and the other not. (The main Obejct with 20% Gyroid-Infill)

On the Layer with Bridge-Infill for the main-Object also the Infill int the massively printed Object is printed from the Material for the Main-Body.

Screenshot of the Type-View:
[image]

Screenshot of the Extruder-View:
[image]

Screenshot of next Layer with correct Extruder:
[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

Version 2.9.0 Flatpak

### Operating system

Manjaro-Linux

### Printer model

Prusa XL - 5 Tool

u89djt, 2025-01-19:
You need to zip the project file to post it.

AshmodaiVirolac, 2025-02-02:
Ahh, yes here is also the Project file:

[Blende-3fach-2-2-Blende-git.3mf.zip](https://github.com/user-attachments/files/18631961/Blende-3fach-2-2-Blende-git.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]