prusa3d/PrusaSlicer #14015: Paint-on supports not respecting `On overhangs only`

JonasVerhofste, 2025-01-20:
### Description of the bug

When setting a highlight by angle in the Paint-on Support tool, the `On overhangs only` toggle of the brush is not respected and also paints non-highlighted parts. This happens with both the sphere and the circle brush.

I would expect the brush here to only fill in the already highlighted parts.

[image]

### Project file & How to reproduce

Grab any file with an overhang and use an overhang highlight + paint `On overhangs only`.

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

Arch Linux

### Printer model

Voron 2.4

u89djt, 2025-01-20:
You've set the overhang angle to 90 degrees, which includes vertical faces. Set it smaller and it'll do what you expect.

neophyl, 2025-01-20:
The error is in the visualiser that is highlighting I think.  90 degrees means vertical walls.  Set something like 89 and you should see that it doesn't paint on the verticals.  That does call into question how accurate the highlighting is.

JonasVerhofste, 2025-01-20:
> You've set the overhang angle to 90 degrees, which includes vertical faces. Set it smaller and it'll do what you expect.

Oh god. I thought I had tested that. 🤦
I guess I mixed two different tests and didn't notice this.

> The error is in the visualiser that is highlighting I think. 90 degrees means vertical walls. Set something like 89 and you should see that it doesn't paint on the verticals. That does call into question how accurate the highlighting is.

89° indeed behaves correctly. I guess fully vertical shells aren't highlighted, because they're unsupportable. (or it doesn't make much sense to do so anyways)

I guess the 0° vs 90° had me confused because I looked at it from a wrong perspective.
Apologies for the overhead.

closed 2025-01-20 by JonasVerhofste