prusa3d/PrusaSlicer #14016: Prusa Connect blank even when logged in

liutang, 2025-01-20:
### Description of the bug

Prusa Connect does not load from PrusaSlicer v2.9. I am logged in. The printables tab loads correctly. However, Prusa Connect tab will flash the spool icon and then followed by a blank screen. I cannot send gcode to the printer too since it cannot find connect and find my printer.

2013 Macbook Pro 15"
Mac OS Big Sur v11.7.10
PrusaSlicer v2.9

[image]

[image]

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.9.0

### Operating system

macOS Big Sur v11.7.10

### Printer model

Prusa i3 MK3S+

tatiang, 2025-01-23:
Sorry, I didn't see this before I posted a similar problem: https://github.com/prusa3d/PrusaSlicer/issues/14028

BlinkSTOP, 2025-01-24:
I think this may be linked - I was logged into PrusaSlicer 2.9.0 on my Win 11 PC, but Send To Connect was coming up with the error "Failed to load a web browser". It was working fine one print earlier.
I tried to re-log but on pressing Log In I get the same error.
I have reinstalled 2.9.0, and rolled back to 2.8.0, but get the same problem.
I have not uninstalled any browsers and my MK3.5S with MMU3 is on the network and remains accessible from Connect online.

BlinkSTOP, 2025-01-24:
initial fault
[image]

trying to log back in
[image]

after rolling back to 2.8.0
[image]

BlinkSTOP, 2025-01-25:
Sorry I just found this reported on #13973 hope there's a fix soon.

liutang, 2025-02-18:
Better, but not quite there yet. It loads a spinning spool.

[image]

kocikdav, 2025-02-18:
@liutang Thanks for quick response, would you please try if reloading does do something? Edit -> Reload Web Content or ctrl+R.

liutang, 2025-02-20:
Nothing happens. The spinning spool reloads.

liutang, 2026-09-18:
This can be closed. I'm unable to replicate with v2.9.6.

closed 2026-09-18 by liutang