prusa3d/PrusaSlicer #14017: Error with Organic Support Default settings for 0.8 nozzle

thew1se, 2025-01-20:
### Description of the bug

When using the default settings for 0.8 nozzle with Organic supports, the following error is shown in prusa slicer:

"ERROR: Organic support tree tip diameter must not be smaller than support material extrusion width"

I have worked around this by changing the tip diameter setting to 0.8.  (Default 0.6)

*Note that the organic supports do not separate off well with PETG / 0.8n / Default settings.

Default for 0.6n is 0.6
Default for 0.4n is 0.6
Default for 0.25n is 0.6

PS: What should this setting be for all other nozzles?

### Project file & How to reproduce

[Computer Buttom Mount.zip](https://github.com/user-attachments/files/18481950/Computer.Buttom.Mount.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.9.0

### Operating system

Windows 11

### Printer model

MK3s+

rtyr, 2025-01-21:
Fixed via configuration update. Thank you.

closed 2025-01-21 by rtyr

RedactedHosting, 2026-01-02:
same issue as of today

rtyr, 2026-01-02:
> same issue as of today

Are you talking about Prusa3D printer with 0.8mm nozzle?