prusa3d/PrusaSlicer #14020: install keeps failing

bushpig7, 2025-01-20:
hello
    just want to say that I will have to think long and hard about continuing to use prusaslicer. installing with flatpak was a horrible experience. what used to take 20min or so took all day and "apparently a 1.5gb download (not even close to the size quoted on your webpage) takes up 3.8 gb of disk space. I dont have a super fast connection and when something goes wrong it means hours of downloading. your appimages "just worked" like your page said. they were of a reasonable size and delivered high quality software. I decided i would try your flatpak delivery system. How bad could it be?' i thought. The new system failed three times before I got it working and every time I was forced to download 1.5 gb . I also noticed at the end when it finally worked that it began snooping into myprinter controller firmware. then filled the console with "61R" over and over again. I have multiple computers that I usually put prusaslicer on. I don't know If I will anymore.

Samueru-sama, 2025-01-21:
Hi, we have an unofficial AppImage you can use [here](https://github.com/prusa3d/PrusaSlicer/issues/13653).

bushpig7, 2025-01-26:
thank you for putting out the appimage, I'm sure lots of people will use it. I think prusa made a mistake with flatpak, Lots of people still use cable connections to printers and have machines devoted to each printer. not many people connect their state of the art computers to their printer. Its nice to have slicing software on those machines but flatpak makes it such an ordeal to install and requires an undesired network connection. I like the stand-alone setup and appimage makes it easy to put software on those machines.

closed 2026-09-29 by prusaslicer-bot[bot]