prusa3d/PrusaSlicer #14021: [ENHANCEMENT] Wipe tower improvement

ekholm, 2025-01-21:
### Printer model

XL 5T

### Describe the enhancement

When using wipe tower the tool head stays in place after a layer is printed and before changing tool.

This delay leads to stringines as the nozzle is stationary for a few seconds. (using PETG).
Everything else is spotless

I think that it would be better if the beed is lowered a bit and tool is moved away from the wipe tower before this delay.

ekholm, 2025-01-24:
> I think this is related to: [#13987](https://github.com/prusa3d/PrusaSlicer/issues/13987)

Yes, it seems so.
thanks

ekholm, 2025-01-24:
Closing it as it's a duplicate

kubispe1, 2025-02-19:
Hi, thanks. I am closing this issue as a duplicate. Please follow up on #13987.

closed 2025-02-19 by kubispe1