prusa3d/PrusaSlicer #14022: PrusaSlicer flatpak on Linux Mint cannot see folders on /mnt or /media/user

valemaio2, 2025-01-21:
### Description of the bug

I have a network share with my 3D printing projects served as a Samba share and mounted via fstab on Linux Mint Mate 21.3. PrusaSlicer installed via flatpak is unable to see any folders inside /mnt or /media/user, despite being owned by the user I'm currently using to log into the system. Creating a soft symlink to the NAS onto my home folder does not work either, as it cannot see the symlink either.

Previous issue resolution was to use the .appimage version of the slicer, but it is not possible anymore as support for it has dropped.

### Project file & How to reproduce

File - Open project
Nothing showing up on /media/user, /mnt is not showing at all.

[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.0 Linux Flatpak

### Operating system

Linux Mint Mate 21.3

### Printer model

MK3s+

SachCZ, 2025-01-21:
Hello, please try:

https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2563593643

valemaio2, 2025-01-22:
> Hello, please try:
> 
> [#13820 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/13820#issuecomment-2563593643)

That did the trick, all I had to do was adding /mnt to the paths available to PrusaSlicer.

[image]

closed 2025-01-22 by valemaio2