prusa3d/PrusaSlicer #14028: Prusa Connect tab is blank, can't select printer to connect to

tatiang, 2025-01-23:
Just recently, I can no longer print to my Prusa MK4S printer using PrusaSlicer 2.9.0. After slicing it, it no longer shows an option for sending the job to Connect.

[image]

And then on the. Prusa Connect tab, it's blank even if I quit and re-open the app or restart my computer. MacBook Air M2 running OS X 15.1. 

[image]

tatiang, 2025-01-23:
See similar issue here: https://github.com/prusa3d/PrusaSlicer/issues/14016

lukasmatena, 2025-02-18:
Can you please retest this in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1)? There is good chance that the problem will be gone, although there are multiple possible causes. Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]