prusa3d/PrusaSlicer #14030: PrusaSlicer 2.9.0 Linux flatpak displays blank Plater screen

tsmaster, 2025-01-23:
[image]

I've been using PrusaSlicer on Linux for several versions, and have been using the new "flatpak" 2.9.0 build for most of January. Today, I tried running it, and got a blank main view (see image above). I went back to 2.8.1, which still works fine.

In between the last time that 2.9.0 ran correctly and now, I believe I upgraded my nVidia CUDA software (auto update) and rebooted.

I tried running PrusaSlicer 2.9.0 with --opengl-debug, but that didn't add any information, any helpful error messages to the console output. I do see a bunch of this message:

18:59:18: Debug: window wxTreeCtrl@0x577b6af5f100 ("treeCtrl") lost focus even though it didn't have it

but I've seen that before, and it seems harmless (if weird).

Jan-Soustruznik, 2025-01-23:
Hello, @tsmaster , A similar issue was addressed here https://github.com/prusa3d/PrusaSlicer/issues/13653#issuecomment-2567132082, that this could be an error related to the graphics driver. Could you check, please? Thank you, have a nice day

DagBjorndal, 2025-02-06:
I have the same problem: 

Prusa Slicer 2.9.0 is unusable on Raspberry Pi 5 Bookworm, because the Plater screen is blank.

Jack-Edwards, 2025-02-13:
Same problem on fully patched Fedora 41.

the-ikletti, 2025-03-03:
> Hello, [@tsmaster](https://github.com/tsmaster) , A similar issue was addressed here [#13653 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/13653#issuecomment-2567132082), that this could be an error related to the graphics driver. Could you check, please? Thank you, have a nice day

Had the same problem as the OP with PrusaScicer 2.9.0 Flatpak after installing the NVIDIA drivers on my Ubuntu 24.04 via `ubuntu-drivers install`. Followed your link.

The info from [this post](https://github.com/prusa3d/PrusaSlicer/issues/13653#issuecomment-2567132082) fixed it:

- Flatpak does not use the installed drivers, instead it uses its own, so you need to install them.

- You need to check your NVIDIA driver version (mine is 550.120):
[image]

- `flatpak install nvidia-550` gives a list of package versions, look for the version matching your driver version.
- select, install, enjoy.

Jack-Edwards, 2025-03-07:
Installing the matching driver version doesn't appear to work either.

[image]

[image]

[image]

rp10007, 2025-04-10:
Another one on Ubuntu 22.04.
2.91 was working a week or two ago, now blank plater screen. All the  other screens  appear to work fine, including Prusalink with the little thumbnails of uploaded designs,  but Plater is just completely blank. 

In addition to the error "Failed to load module "canberra-gtk-module"", there is  also another error "libEGL warning: egl: failed to create dri2 screen"

and preceding that a series  of notifications:

```
20:41:04: Debug: window wxTreeCtrl@0x5eb875fb9e90 ("treeCtrl") lost focus even though it didn't have it
20:41:04: Debug: window wxTreeCtrl@0x5eb875fb9e90 ("treeCtrl") lost focus even though it didn't have it
20:41:10: Debug: window wxTreeCtrl@0x5eb875fb9e90 ("treeCtrl") lost focus even though it didn't have it
20:41:10: Debug: window wxTreeCtrl@0x5eb875fb9e90 ("treeCtrl") lost focus even though it didn't have it
```

In addition, quitting by menu item doesn't  appear to  actually quit. Closes the PS window, but takes me back to the  terminal window with error  messages, and I have to ^C to get back to a prompt.

Oh, and one more thing: on startup the terminal window also show a long list of error messages of the form 

`"[2025-04-09 20:59:53.777545] [0x00007ac166b08e40] [error]   Error in a preset file: The preset "/home/xxxx/.var/app/com.prusa3d.PrusaSlicer/config/PrusaSlicer/print/plus1 1.1utility2strongbrim.ini" contains the following incorrect keys: extruder_clearance_height, extruder_clearance_radius, which were removed"`

followed by 

"[2025-04-09 20:59:54.095432] [0x00007ac166b08e40] [error]   UserAccount: Failed to read token - no datafile found."

Those had not afaik been appearing before, so I'm wondering whether the .ini file somehow got  messed up.

rp10007, 2025-04-10:
OK, followed the discussion trails, and removal/reinstallation appears to have fixed things. Only one minor mostly unrelated nit: The update check says that 2.92  is ready, but flatpak unqualified install gives me 2.91. I guess propagation to flathub is not quite in sync with release of other flavors.

SachCZ, 2025-04-14:
Hello @rp10007, glad to hear the re-installation helped. Flathub is about 6-8 hours behind the release on Github.

barrenaedu, 2026-09-24:
'm having same no bed plate error on latest debian + wayland:

```
Prusa Slicer Version 2.9.6
Linux 6.12.107+deb1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.107-1 (2026-08-29) x86_64 GNU/Linux
NVIDIA GeForce RTX 3080 Lite Hash Rate [Discrete]  Driver Version: 595.104.02  
Debian GNU/Linux 13 (trixie) x86_64
KDE Plasma 6.3.6
KWin (Wayland)
Theme: Breeze (Light) [Qt], Breeze [GTK2/3]
---
$ flatpak run com.prusa3d.PrusaSlicer
--------------------------------------------------------------------------
Message: 11:14:32: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
Gtk-Message: 11:14:33.004: Failed to load module "colorreload-gtk-module"
Gtk-Message: 11:14:33.004: Failed to load module "window-decorations-gtk-module"
11:14:36: Debug: window wxTreeCtrl@0x560ec7439290 ("treeCtrl") lost focus even though it didn't have it
11:14:36: Debug: window wxTreeCtrl@0x560ec7439290 ("treeCtrl") lost focus even though it didn't have it
11:14:41: Debug: window wxTreeCtrl@0x560ec7439290 ("treeCtrl") lost focus even though it didn't have it

```

jabdoa2, 2026-10-26:
Same issue on Lubuntu 24.04. This seems to sometimes resolve after a few minutes. But other instances never show proper windows. I guess this is either blocking on something or a race.

```
Version: 2.9.6
Build: PrusaSlicer-2.9.6+flathub.org
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 7.0.0-31-generic x86_64
Total RAM size [MB]: 134,324MB

Resident memory: 2,027MB
Shared memory: 274MB
Private memory: 1,753MB
Peak memory usage: 2,258MB
RAM size reserved for the Undo / Redo stack: 1,074MB
RAM size occupied by the Undo / Redo stack: 2MB
OpenGL installation
GL version: 4.6.0 (4.6 (Core Profile) Mesa 26.2.2 (git-3281a69a8b))
Profile: Core
Vendor: AMD
Renderer: AMD Radeon Graphics (radeonsi, renoir, ACO, DRM 3.64, 7.0.0-31-generic)
GLSL version: 4.60.0
Textures compression: Enabled
Mutisampling: Enabled (4 samples)
Installed extensions:
GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_gpu_shader_int64
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
[... 173 lines cut ...]
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_shader_subgroup
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_framebuffer_flip_y
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_const_bandwidth
GL_MESA_texture_signed_rgba
GL_NVX_gpu_memory_info
GL_NV_alpha_to_coverage_dither_control
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_packed_depth_stencil
GL_NV_shader_atomic_int64
GL_NV_texture_barrier
GL_NV_timeline_semaphore
GL_OES_EGL_image
GL_S3_s3tc
Eigen vectorization supported: SSE, SSE2
```