prusa3d/PrusaSlicer #14035: Add option to disable the colour change button on the layer selection bar

ygoe, 2025-01-23:
**Is your feature request related to a problem? Please describe.**
In the slicing preview, the layer selector bar (similar to a scrollbar) has the special behaviour that just clicking the thumb without dragging it adds a colour change. No normal scrollbar would ever do anything when clicking (not dragging) its thumb. I accidently clicked that thing while slowly browsing through the layers and not moving the thumb far enough to go to the next layer. When that happens, the view changes drastically (I guess it falls out of slicing preview) and a separate compensation action is required to undo the unintended change.

**Describe the solution you'd like**
I'd like to have an application setting that allows disabling that button click behaviour. My printer can't change colours during a print and I'm not going to need or do that manually anytime soon.

**Describe how it would work**
The user activates that application setting and then the plus icon would disappear from the layer selector thumb and clicking without dragging it has no effect anymore.

**Describe alternatives you've considered**
I could be more careful how to interact with PrusaSlicer's UI as it's a little dangerous at that point. I will make mistakes and be annoyed. The UI has sharp edges and I will cut myself again.

aimbot-luc, 2025-01-23:
But like you said it is not a traditional scrollbar, it's designed to preview the layers and also add extra functionality such as adding custom g-code on custom heights. And (imo) it works great for that purpose.

If you want to scroll slowly through the layers without accidentally pressing the 'color change/M600' button you can also use the scroll wheel of your mouse while hovering over the bar, use the up/down arrows or even the **w** (up) / **s** (down) keys on your keyboard. This will move through the layers one at a time.

And if it happens that you clicked the button by accident, you will see the colors change and also see a 'revert' arrow on top of the bar to remove all custom layer additions. See below.

[image]

neophyl, 2025-01-24:
If your printer can’t do a colour change and you want to disable that functionality on the scroll bar then just go into your printer profile and delete the gcode in the colour change field. Probably will be M600. Then save the change to the profile. 
That way no more accidental colour change additions

ygoe, 2025-02-09:
> If your printer can’t do a colour change and you want to disable that functionality on the scroll bar then just go into your printer profile and delete the gcode in the colour change field. Probably [...]

I have no idea what you mean. My printer is the Prusa Mini+, and Prusa (who probably made the profile) should already be aware that this model can't change colours.

I've watched things more carefully now. When pressing that "button" handle, no dialogue appears. I just fall out of slicing mode. That changes the 3D model view by changing its colours (e.g. from orange to yellow) and shows the (normally invisible) layers above the current layer half-transparently. The app forgets what it has done and brings me back to the model view. I have to do two separate things to continue browsing the layers:

1. Clear that colour change by scrolling back to the layer where it was inadvertently added, then click the "x" button that appears next to the scrollbar
2. Click "Slice now" again and wait for the operation to complete

That behaviour is a bug in my opinion and should be fixed.

closed 2026-09-29 by prusaslicer-bot[bot]