prusa3d/PrusaSlicer #14036: Alternating print order of multiple objects

MalteBrunn, 2025-01-24:
**Is your feature request related to a problem? Please describe.**
When printing multiple objects the order each object is printed at is the same for each layer. This increases the number of travel moves at the layer change. Thus, increasing print time.
I assume this problem depends on the object shape. It was most notably for me while printing many longish objects.

**Describe the solution you'd like**
Alternating the order the objects are printed each layer reduces travel moves, as the hotend will zig-zag around the plate and does not have to return all the way before starting the new layer.

**Describe how it would work**
For simplicity assume we have to objects (A and B) on the build plate. On layer X the objects are printed A first, B second. On layer X+1 the order is reversed. Thus, B first, A second. On layer X+2 the order is the same as for layer X and so forth.

**Describe alternatives you've considered**
None. But the solution rendered seems quite obvious to me.

**Additional context**
A similar problem appears while printing many instances of the same object (again with longish objects like a ruler) but within the layer for travel moves between instances. This is because all instances start at the same location. Using hard copies of the object instead of instances helps, as they are processed individually. Heaving two independently processed Gcode patterns (a and b) for a set of instances (e.g. 1, 2 ,3 ,4) and alternating between them would help. E.g., 1a -> 2b -> 3a -> 4b on layer X and on X+1 4b -> 3a -> 2b -> 1a. However, I consider this a different problem as it is more involved.

u89djt, 2025-01-24:
You need to zip the project file to post it.

MalteBrunn, 2025-01-24:
I just created a simple mock up with regular boxes. The long diagonal travel move from the first to last object is what I'm referring to. The shorter diagonal moves between each box is what I mentioned in the "Additional context". 

The long moves could easily be avoided be printing each other layer in the reverse object order.

[boxes.zip](https://github.com/user-attachments/files/18538743/boxes.zip)

[image]

u89djt, 2025-01-24:
If you switch to "nearest" seams, you get this, which still prints in the same object order, but you haven't asked it to do everything the same way so there's less flying around within the object. I wonder if there's something about parts cooling sufficiently? If you start extruding immediately on top of the end of a perimeter, I should imagine that could move more than is desirable. Anyway, the devs have got their reference file now, so over to them or someone who already knows why this logic is in place or happens that way.

[image]

MalteBrunn, 2025-01-24:
Yes, using nearest seem position helps. but there is still the big travel move from last to first object. Cooling should not be much of an issue I guess as each object prints on its own just fine.

For completeness: I tested with 2.8.1+linux-x64-GTK3 (AppImage) and 2.9.0 (Flathub)

u89djt, 2025-01-24:
Ah yeah, so it would have to be something about uneven cooling across the objects. The end two might have the last two layers wriggling a bit differently. I'm jousting with plastic shrinkage at the moment trying to make really accurate curves. There's a lot going on.

Tupson444, 2026-09-19:
still relevant