prusa3d/PrusaSlicer #14038: Cross-compiling documentation

LaurensKrudde, 2025-01-24:
**Is your feature request related to a problem? Please describe.**
The current documentation for building PrusaSlicer is limited and covers only a standard build. However, it appear that the cmake files also account for cross-compilation setups. Also, there are some cmake presets that are not documented.

**Describe the solution you'd like**
A more elaborate documentation on the current build/cross-compilation possibilities would be helpful.

LaurensKrudde, 2025-01-27:
In my case, I'd like to build for a Raspberry Pi 5, which uses an 64-bit ARM processor (aarch64) and runs Linux. I have a computer with a x86_64 processor and run Linux using WSL. I need to cross-compile. Is that possible? And if so, is there documentation available on how that is done?

u89djt, 2025-01-27:
They're explicit about what they support in the repository in the readme. Familiarising yourself with compilation tools and methods doesn't happen in the bugs and features report system. But, you never know, someone watching might have tried to jam 2.9 into a 5 with tales to tell. Googling turned up one character dismayed that it's slower than 2.8, so there are people out there trying.

closed 2026-09-29 by prusaslicer-bot[bot]