prusa3d/PrusaSlicer #14041: FR: Build plates - Add plate number to the filename for unique file names

RPhilbo, 2025-01-25:
**Is your feature request related to a problem? Please describe.**
When using multiple print beds in a project, the exported file name is always the same. Resulting in overwriting the former gcode file from plate n-1.

**Describe the solution you'd like**
It would be more comfortable, to add a prefix/postfix to the filename, when number of build plates is > 1.

**Describe how it would work**
- Check if number of build plates is > 1.
- If so, then add a textframe like "_build-plate-n" with n equals the build plate number. That text may be attached prior or past the actual file naming.

**Describe alternatives you've considered**
Doing it by hand for every build plate.

**Additional context**
Example:
A: 01_Base_assembly_9h54m_0.28mm_230C_PET_ENDER3V2.gcode
B: 01_Base_assembly_9h54m_0.28mm_230C_PET_ENDER3V2_build-plate-3.gcode

stixpjr, 2025-02-16:
Just started experimenting with having multiple beds, and was surprised to find the sliced gcode for each bed has the same generated name. So +1 to this, although I'd be happier with a more succinct "_bed1_", etc.

MrTrick, 2025-05-17:
Is this feature not already present? I was just commenting on it in #14231.

RPhilbo, 2025-05-18:
Well your ping sent me to update from 2.9.0 to 2.9.2, but it is still the same filename for all plates/beds.

[image]

MrTrick, 2025-05-18:
@RPhilbo if you have multiple beds and open the preview pane, there will be a "Slice all" square to the left of the numbered beds. 

If you click on that, then the export button should change to say "Export all G-codes..."
And that should add the bed numbers to the file name.

Shinomorisama, 2025-05-19:
Just commenting that I support this feature request.  Even if it adds the bed when exporting all. It would be nice to be able to export only one when working on a project.    Will also throw in that, having used the multiple export many times, waiting for it to scroll to the end to find the right one is annoying.  Would be much nicer to have the bed# right after the file name as mentioned in this request #14231

closed 2026-09-29 by prusaslicer-bot[bot]